InfiniTimeOrg/InfiniTime #1185: Support for Golang/Zig/other for building apps

devnoname120, 2022-06-18:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

C++ for building apps is hard for many users. I suggest to add support for a compiled language that is fast and with a low runtime footprint.

### Description

MicroPython could be an option but the downside is that it would slow down the otherwise super snappy and battery-saving InfiniTime OS.

devnoname120, 2022-06-19:
**Zig in particular has**:
- Low memory/CPU footprint
- First-class support for calling C/C++ code.
- Supports C/C++ compilation out of the box so we could switch the whole project to use their compiler and make Zig compilation/integration seemless.

Arsen6331, 2022-06-19:
Theoretically, this should already be possible, though it would probably be more difficult than just writing the C++ itself. I'd wait for support for uploading apps, which is planned sometime in the future, and then see how those are implemented and do the same in another language.

Arsen6331, 2022-06-19:
And MicroPython is not really an option. InfiniTime is already running out of RAM as it is, and every new feature uses more RAM. MicroPython would use a lot of RAM and be much slower. If you want MicroPython apps, you can use [WaspOS](https://github.com/daniel-thompson/wasp-os)

JF002, 2022-06-19:
I agree with @Arsen6331, I think we have many other challenges to overcome before adding more possibilities to build apps. We are already working (experimenting) on offloading resources to the external memory in #321.

devnoname120, 2022-06-19:
> I agree with @Arsen6331, I think we have many other challenges to overcome before adding more possibilities to build apps. We are already working (experimenting) on offloading resources to the external memory in #321.

Thank you. Are there plans to merge PineTime Lite changes upstream? 🙏
The UI/UX looks more sleek, and even horizontal scrolling is smoother/faster than InfiniTime.

Riksu9000, 2022-06-20:
Closing as not planned for the time being.

closed 2022-06-20 by Riksu9000

devnoname120, 2022-06-20:
> Closing as not planned for the time being.

I planned to do it myself, not rely on the maintainers. Given your comments I'll wait a bit however.

Riksu9000, 2022-06-20:
I could've elaborated on this a bit more. Mainly I think that core InfiniTime apps should just be written in C++ and it's not worth bringing in more dependencies. We aren't in need of more app developers right now either, since there's not that much memory to spare.