InfiniTimeOrg/InfiniTime #1186: Increase screen drawing

devnoname120, 2022-06-19:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Horizontal animations are incredibly slow

### Reproduction steps

- Run a horizontal animation. Example: https://github.com/InfiniTimeOrg/InfiniTime/pull/1173
- It's laggy and there is some annoying screen tearing.

### More details?

**Example**: horizontal sliding animations. Currently it's super laggy and unnatural. See https://github.com/InfiniTimeOrg/InfiniTime/pull/1173.

See [working implementation for Oxide OS](https://artemis.sh/2022/03/28/oxide-hubris-on-pinetime.html#:~:text=Compare%20these%20two%20videos%20of%20writing%20a%20solid%20block%20of%20color%20to%20the%20screen%2C%20first%20through%20the%20SPI%20task%2C%20and%20second%20with%20direct%20SPI%20hardware%20access).

### Version

95ff285

### Companion app

Gadgetbridge 0.67.1

Riksu9000, 2022-06-19:
The video is of vertical scrolling with the watch turned sideways. The limiting factor of screen updating is the SPI interface. There is a somewhat hacky way to increase the speed further #806.

closed 2022-06-19 by Riksu9000

devnoname120, 2022-06-19:
@Riksu9000 My bad, thanks for your correction. p🙏

devnoname120, 2022-06-19:
@Riksu9000 Did someone try to send command `36h` to switch columns and rows and see if vertical scrolling becomes horizontal scrolling in this mode?

https://www.hpinfotech.ro/ILI9340.pdf#page=127

JF002, 2022-06-19:
I haven't tried to switch the orientation of the display for the animation yet. I assume that we'll need to refresh the whole display twice and that the result won't be as fast as expected, but I might be wrong. As you said : it is worth to be tested!

devnoname120, 2022-06-26:
> I haven't tried to switch the orientation of the display for the animation yet. I assume that we'll need to refresh the whole display twice and that the result won't be as fast as expected, but I might be wrong. As you said : it is worth to be tested!

@JF002 I can confirm that vertical scrolling becomes horizontal scrolling when setting the MV bit to `1`.

I modified a function of the driver like this:

```cpp
void St7789::MemoryDataAccessControl() {
  WriteCommand(static_cast<uint8_t>(Commands::MemoryDataAccessControl));
  WriteData(0x1 << 5u | 0x1 << 7u); // MY = 1, MV = 1
}
```

See this extract from the datasheet of a similar screen for the meaning of these bits:
[image]

**Demo**:

https://user-images.githubusercontent.com/2824100/175830601-328bce58-d662-4fe8-b687-fbe739815e79.mp4

It looks garbled because I didn't change how InfiniTime draws the screen. But I think that it can be solved:
- For vertical scrolling, set `MV` and `MY` to `0` and draw the screen normally.
- For horizontal scrolling, can set `MV` and `MY` to `1` and draw the screen by columns. I'm not sure how much change it would require to InfiniTime.

I'm not sure whether we'll actually need to redraw the screen after setting `MV` and `MY` to `0`/`1`.