InfiniTimeOrg/InfiniTime #1188: Weather Radar

JM480, 2022-06-19:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Weather radars can be more helpful when forecasts are inaccurate.

### Description

The National Weather Service can be used for the US. Maybe the location could be selected in the settings.

JM480, 2022-06-20:
> Can you add an illustration of what you mean?

Something like this app. https://github.com/dh4/WeatherRadar

ItzSwirlz, 2022-06-24:
It would be way too complicated. There needs to be consistent, and I mean consistently data polling

Avamander, 2022-06-24:
I guess a low-res bitmap could be transmitted over BLE, but I don't think the current space constraints make it possible to include this in base IT for now. Probably a thing for the time where we have apps loadable from flash.

closed 2022-08-03 by JM480