InfiniTimeOrg/InfiniTime #1189: date/tz.h can not be impoted foir the lack of pthread

callingmybluff, 2022-06-20:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

tz.h error: 'once_flag' is not a member of 'std'

### What should happen instead?

I should be able to include that 

### Reproduction steps

From the latest dev, try to add these 2 lines to any of your .cpp files:
```cpp
#include <date/tz.h>

auto x = date::current_zone();
```
Then you will get this error while trying to build: 

> /workspaces/InfiniTime/src/libs/date/includes/date/tz.h:792:26: error: 'once_flag' is not a member of 'std'
[build]   792 |     std::unique_ptr<std::once_flag>      adjusted_;
[build]       |                          ^~~~~~~~~
[build] /workspaces/InfiniTime/src/libs/date/includes/date/tz.h:117:1: note: 'std::once_flag' is defined in header '<mutex>'; did you forget to '#include <mutex>'?
[build]   116 | #include <cassert>
[build]   +++ |+#include <mutex>
[build]   117 | #include <chrono>
[build] /workspaces/InfiniTime/src/libs/date/includes/date/tz.h:792:26: error: 'once_flag' is not a member of 'std'
[build]   792 |     std::unique_ptr<std::once_flag>      adjusted_;
[build]       |                          ^~~~~~~~~
[build] /workspaces/InfiniTime/src/libs/date/includes/date/tz.h:792:26: note: 'std::once_flag' is defined in header '<mutex>'; did you forget to '#include <mutex>'?
[build] /workspaces/InfiniTime/src/libs/date/includes/date/tz.h:792:35: error: template argument 1 is invalid
[build]   792 |     std::unique_ptr<std::once_flag>      adjusted_;
[build]       |                                   ^
[build] /workspaces/InfiniTime/src/libs/date/includes/date/tz.h:792:35: error: template argument 2 is invalid

### Version

develop

Avamander, 2022-06-20:
Works as it can and should. Closing.

closed 2022-06-20 by Avamander

callingmybluff, 2022-06-20:
@Avamander, can you please elaborate on how else one can get time zones? or the current time zone?

Avamander, 2022-06-20:
You can't.

JF002, 2022-07-08:
Just to give a bit more detail about this issue : if tz.h depends on pthread, it won't work on InfiniTime. InfiniTime is based on FreeRTOS, which, as most RTOS design for embedded systems, do not support POSIX threads. To work around this, you'll probably want to edit tz.h so it does not depend on pthread or implement timezone without tz.h.