InfiniTimeOrg/InfiniTime #1211: Priority Notifications

Fidgetcetera, 2022-06-29:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

API support for a companion app (ie gadgetbridge) to tell the pinetime if a notification was of a higher priority, and a different kind of vibration for these.

### Description

I have a specific notification sound set on my phone for Android's Priority conversations, and I'd love if (someday, if gadgetbridge implements support for it, etc etc) my Pinetime could know when a notification is higher priority *too*, and buzz differently. That way, I know without looking that one of my partners or close friends is who messaged me, and not that I just got an unimportant github email.

minacode, 2022-06-30:
As far as I understand it, this is not supported by the AlertNotificationService standard (BLE) so it would have to be encoded in the notification payload.

ajack2001my, 2022-07-05:
My question is, what constitutes a priority notification? How is it weighted and will everyone agree or can it be customized?

Fidgetcetera, 2022-07-05:
I'm not requesting weighting - only different notification vibration patterns.