InfiniTimeOrg/InfiniTime #1215: Show heart rate %

rimugu, 2022-06-30:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Calculate % based on age and display %

### Description

Calculate % based on age (maybe from gadgetbridge) and display %
Training with bpm is many times done with zones based on %.
One basic calculation of of is %=bpm/(220-age)*100
Would require rework on font size to make space.

coderamm, 2022-07-27:
I think BPM is a much more neutral way of reporting heart rate. Is there a specific reason you are interested in displaying HR as a %? Is this something other smartwatches do?

rimugu, 2022-07-28:
> there a specific reason you are interested in displaying HR as a %?

Some sports training is done by hearth rate zones (defined as %). Displaying that would be useful for those wishing to use it for formal exercise.
For example: [https://www.polar.com/blog/running-heart-rate-zones-basics/](https://www.polar.com/blog/running-heart-rate-zones-basics/)

mashuptwice, 2022-07-28:
I don't think that with the mediocre PPG sensor of the pinetime it will be possible to get an accurate HR reading during sports.