InfiniTimeOrg/InfiniTime #1217: Meaning of L_HR in terminal watch face?

arduino-man, 2022-07-01:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I don't understand what L_HR stands for

### What should happen instead?

Should say HR only maybe?

### Reproduction steps

I do not understand what the lettler "L" stands for in "L_HR" when using the terminal watch face..

I know HR means heart rate but what is that "L" about?

Thanks

### More details?

No more details

### Version

1.10

apilat, 2022-07-01:
"HR" only has two letters so it wouldn't line up with the rest of the UI. For what it's worth, I personally use "HBPM".
I would also change "STAT" to something that indicated we mean *Bluetooth* status - for example "BLTH".

arduino-man, 2022-07-01:
Fair enough but what does the “L” in “L_HR” stand for?

arduino-man, 2022-07-04:
Seriously, how can this be admitted into the main branch without nobody knowing what that "L" stands for? I know it's weird, but doesn't anyone care? lol.

Avamander, 2022-07-04:
L as in last measurement. Not a very consequential letter in the end.

closed 2022-07-04 by Avamander

arduino-man, 2022-07-04:
Sorry for reopening this..

I've reached out to the original creator of the watch face ZephyrLabs and can post his response as to the meaning of that "L" shortly as I take the above comment: "L as in last measurement" is only a guess?

Cheers

arduino-man, 2022-07-04:
LOL turns out the guy who submitted the watch face to InfiniTime simply copied it from someone else's design and has no idea what that "L" stands for.
Picture:
https://user-images.githubusercontent.com/27705324/177205197-b3e8376a-cf99-4faa-9095-9ec3696f2b9a.jpg