InfiniTimeOrg/InfiniTime #1221: Companion app for Windows PCs to flash Infinitime onto a Pinetime smartwatch

Danimations, 2022-07-02:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

A simple Windows companion app to flash firmware onto Pinetime

### Description

It would be great if someone with coding skills would make a simple companion app for Windows, that would in the very least, allow for easy flashing of Infinitime to a Pinetime smartwatch via Bluetooth. I find that Gadgetbridge does not connect to the Pinetime on some devices, and Android and Windows are both very popular OSes. Being able to at least update the OS from a Windows PC would make Pinetime more user friendly and help expand the potential user base of Pinetime smartwatches.

ITCactus, 2022-07-04:
not sure if it still works, but [this Wiki page](https://wiki.pine64.org/wiki/PineTime#Companion_Apps) claims:
> [PinetimeFlasher](https://github.com/ZephyrLabs/PinetimeFlasher) (Windows) - Companion desktop app, only supports flashing firmware.

however it uses "openOCD", so it's not for Bluetooth and sealed devices...

the first thought was about nRF tools for windows, but still [not supported](http://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/matter/nrfconnect_examples_software_update.html?highlight=device%20firmware%20upgrade#device-firmware-upgrade-over-bluetooth-le-using-pc-command-line-tool)...

> On Windows, there is no support for Device Firmware Upgrade over Bluetooth LE yet.

ITCactus, 2022-07-04:
maybe, as an alternative, the web-page that uses [Web Bluetooth](https://googlechrome.github.io/samples/web-bluetooth/) could be used, and that should work [not only on windows](https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md)

JF002, 2022-07-04:
I haven't heard about any development efforts on Windows yet, indeed!
For those interested to take the challenge : [Siglo](https://github.com/theironrobin/siglo) is written in Python and [ITD](https://gitea.arsenm.dev/Arsen6331/itd) in Go, and might be not too difficult to port on Windows.

ItzSwirlz, 2022-07-04:
In general, for desktop (aside from Linux) there is no application. I guess maybe I'll look at porting Siglo one day but I wouldn't expect much as I don't know about Mac or windows system structure with Bluetooth

Danimations, 2022-07-17:
> maybe, as an alternative, the web-page that uses [Web Bluetooth](https://googlechrome.github.io/samples/web-bluetooth/) could be used, and that should work [not only on windows](https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md)

This seems like a smart approach, which could lead us to a universal solution! I hope someone reading this decides to explore this possibility.

JF002, 2022-07-19:
Yes, indeed, the developer behind WebBLEWatch created this web app as a proof of concept, and it works very well (only under Chrome, iirc). @joaquimorg from [PineTimeLite](https://github.com/joaquimorg/PinetimeLite) has also created a [web app](https://home.joaquim.org/pinetime/) for their fork!

Arsen6331, 2022-09-23:
> I haven't heard about any development efforts on Windows yet, indeed! For those interested to take the challenge : [Siglo](https://github.com/theironrobin/siglo) is written in Python and [ITD](https://gitea.arsenm.dev/Arsen6331/itd) in Go, and might be not too difficult to port on Windows.

The issue with trying to port existing Linux companions to Windows is that their bluetooth stacks and the way a program has to interact with the bluetooth stack are completely different. Same applies to macOS. ITD compiles for Windows and macOS, and would run fine on them, but there is no DBus and no BlueZ on Windows, and that's what ITD, Siglo, and all other Linux companions will be trying to interact with. Additionally, a lot of ITD's features are implemented via DBus services. For example, the call notifications come from ModemManager which is a Linux program that exposes a service on DBus. Regular notifications come from a freedesktop DBus standard used by all Linux programs, but not on Windows. Even music controls use a DBus standard called MPRIS.

A companion for Windows would most likely have to be written from scratch because Windows is very different from Linux.

The way this is handled in WebBLE is via an abstraction for Bluetooth stacks that is only present in Chrome currently (not Firefox, which is what I use). Unfortunately, the closest thing that exists to that in Go is https://github.com/tinygo-org/bluetooth, and it doesn't have enough functionality on Windows and macOS, and even if it did, that wouldn't solve all the other issues.

TailyFair, 2023-01-25:
I tried all other ways to update firmware via BLE on windows, using WSL2, Ubuntu VMs are none of the solution worked.
So I started creating prototype companion app for windows:
https://github.com/TailyFair/InfiniWindows

Currently only Firmware update is working via BLE (without resources). But it works and I was able to flash my own firmware builds. 

Hopefully this won't count as promotion, just sharing to lower the entry barrier for new contributors.

JF002, 2023-01-28:
@TailyFair Thanks for sharing your project with the community! I'm excited to see some work done on a companion app running on Windows! I'll keep an eye on your project and test it as soon as I'll reboot my computer on Windows :)

ProgramminCat, 2023-09-07:
I started making a companion app for Windows (and Linux, MacOS)
https://codeberg.org/InfiniTime-Tools/InfiniConnect
Notifications are working and i plan to add more features

JF002, 2023-09-09:
> I started making a companion app for Windows (and Linux, MacOS) https://codeberg.org/InfiniTime-Tools/InfiniConnect Notifications are working and i plan to add more features

Awesome! I'll keep an eye on your repo on Codeberg!