InfiniTimeOrg/InfiniTime #1224: Release a test build whenever a new change is approved in milestones...

ajack2001my, 2022-07-05:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Release test builds for latest completed change/update in milestone until new firmware is released...

### Description

Whenever a PR is approved (as listed in milestones), a new test build is released (labelled as test eg. InfiniTime-1.11.0-test_20220705) until all the PR in a milestone is complete... That way others can test and report back issues while waiting for next final milestone build.

Riksu9000, 2022-07-05:
Every PR is built by GitHub actions and the DFU is uploaded. The same happens each time the develop branch is updated. It's not the easiest to find though. I'm for creating nightly builds and making them more accessible, but I wouldn't want to clutter the releases with them, so I wonder where they should be put.

ajack2001my, 2022-07-05:
@Riksu9000 thanks for the quick reply. Looking forward to you updating us where we can access the nightly builds (when you create them) and making them easily accessible. :)

ajack2001my, 2022-07-07:
@Riksu9000, I downloaded the DFU .zip file but couldn't send the file to my PineTime... Can I ask why?

JF002, 2022-07-08:
@ajack2001my Which companion app do you use? Most of them accept the .zip file as input, but I think Amazfish only supports the .bin file (extracted from the DFU zip file).

ajack2001my, 2022-07-09:
I use Android/GB to push the build

closed 2025-06-28 by mark9064