InfiniTimeOrg/InfiniTime #1232: swipe to remove new notification

mashuptwice, 2022-07-11:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

new notification dismissable by swipe while preview is shown

### Description

While dismissing notifications works fine while in the notification screen, I would like to be able to instantly dismiss a new notification during the time it gets shown on the display and (if possible) also return to the previous screen. 

Currently to dismiss a notification which just came in, I have to:
0. receive notification
1. swipe up to show home
2. swipe down to show notifications 
3. swipe right to dismiss the notification
4. swipe up to show home
5. (navigate to previous app)

My proposal would make it as following:
0. receive notification
1. swipe right to dismiss
2. (navigate to previous app)

devnoname120, 2022-07-12:
cc @neoburner

SlidingHorn, 2022-11-22:
Hey all - I'm just curious if this is planned to make it into 1.12 - it's a feature I'd love to have, as it's a pain to have to make extraneous swipes (down, then up, then down, then right) rather than being able to dismiss the new notification outright. I see that originally #1245 was created and closed in favor of #1332, which (as far as my non-developer eyes can see) looks ready to merge.

Any updates or insight?

Thanks in advance, and thanks for all the hard work you all do here. Love my PineTime and Infinitime!

devnoname120, 2022-11-22:
I've also been using this PR for a while without any issue.

NeroBurner, 2022-11-29:
@SlidingHorn from my point of view, yes. I'd like to see this improvement in 1.12. The limiting factor is time, especially from the maintainers, as we all do InfiniTime in our free time. But I think (as I've seen no road-block) nothing else is blocking this feature

Riksu9000, 2023-01-14:
This has been implemented with #1332 and #1350

closed 2023-01-14 by Riksu9000