InfiniTimeOrg/InfiniTime #1238: Updating Pinetime Watch using nRF connect causes watch to reboot

Column01, 2022-07-13:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When trying to update my pinetime to the latest firmware OTA using nRF connect, it disconnects from bluetooth andd reboots

### What should happen instead?

It should try and actually update the firmware

### Reproduction steps

1. Watch is running firmware 1.6.0
2. nRF is the latest version from the playstore
3. Connect to watch
4. Click "DFU" button
5. Select the Distribution packet option
6. Select latest firmware (1.10.0 release)
7. Watch shows firmware update and the upload starts but makes no progress (sometimes will jump to 13% at like 130KB/s)
8. Watch reboots and nRF connect tries to reconnect and triggers the update again causing it to reboot again

### More details?

Not sure what to do, I'm going to try updating to an older firmware first and see if it gets resolved doing that.

### Version

v1.6.0

### Companion app

nRF Connect

Column01, 2022-07-13:
It appears to be working using Gadgetbridge so this is probably an nRF connect bug?

Column01, 2022-07-13:
Yup was able to get the latest FW using Gadgetbridge

SteveAmor, 2022-07-14:
This is a known problem with nRF Connect, not InfiniTime. Linked to #1066.

JF002, 2022-07-19:
It's a known issue : newer versions of NRFConnect won't work anymore with the PineTime. NRF probably implemented changes that are are compatible with InfiniTime. 
Note that NRFConnect is a **closed source** app developped by NRF. At the beginning of InfiniTime, NRFConnect was convenient as InfiniTime implements the same "legacy" DFU protocol than NRFConnect. Since then, we have many alternative companion apps, and I would not recommend on using NRFConnect anymore. On Android, you can use [Gadgetbridge](https://f-droid.org/app/nodomain.freeyourgadget.gadgetbridge)

mormegil-cz, 2023-07-13:
While I agree it is not the job of InfiniTime to suppot NRFConnect, I want to point out the current documentation still [claims NRFConnect is one of two supported apps to update PineTime](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/gettingStarted/updating-software.md#updating-with-companion-apps) with [a specific guide how to do so](https://github.com/InfiniTimeOrg/InfiniTime/blob/main/doc/gettingStarted/ota-nrfconnect.md).

Which I just followed and found it does not work, and arrived finally here. So, the documentation should be updated at least to point this out (or the NRFConnect mentions removed altogether?), I’d say.

(Also, note that Gadgetbridge currently [does not (yet) support resource update](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2919), so using it to upgrade InfiniTime is now problematic as well.)

JF002, 2023-08-06:
@mormegil-cz I was wondering why some users would still use NRFConnect while it's know for quite a long time that it's not working anymore with InfiniTime.
We should update the documentation ASAP and remove all mentions to NRFConnect!

> (Also, note that Gadgetbridge currently [does not (yet) support resource update](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2919), so using it to upgrade InfiniTime is now problematic as well.)

Resource update uses a custom protocol that was implemented specifically for InfiniTime, so NRFConnect has never supported this feature as well. I also hope that Gadgetbridge, InfiniLink and other companions apps will implement resource update in the near future, though.

JF002, 2023-08-13:
This PR updates the documentation and removes all mentions to NRFConnect : https://github.com/InfiniTimeOrg/InfiniTime/pull/1827