InfiniTimeOrg/InfiniTime #1240: Stopwatch Button Placement

RageGamerBoi, 2022-07-19:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Widen the start/pause and reset/lap buttons, and move the reset/lap button to the top of the screen.

### Description

Similar to the new timer UI, but with the stopwatch.

Something like this would make accidental button misclicks a lot less likely (and the stopwatch nicer):

[image]

Riksu9000, 2022-07-21:
Buttons should generally be on the bottom of the screen so the finger isn't in covering the screen, and missclicks aren't an issue with the buttons as far as I'm aware.

closed 2022-07-21 by Riksu9000

RageGamerBoi, 2022-07-21:
> Buttons should generally be on the bottom of the screen so the finger isn't covering the screen

Typically when hitting the reset/lap button, one isn't looking at the stopwatch and/or doesn't need to see it, and if they are, they are more then likely looking at the stopwatch to see exactly where the lap or reset button is.

The main purpose of separating and widening the buttons is to prevent misclicks (this would have been somewhat mitigated by PR #558, but this is a much better and less invasive solution)

> missclicks aren't an issue with the buttons as far as I'm aware.

With the stopwatch especially, misclicks are often the result of buttons that are too small or improperly spaced. The current buttons are only about a half-inch wide, while average human thumb is around about 1 inch wide, making it much easier to misclick when hitting the lap button while doing physical activity (where a stopwatch is often used), because our arms move a lot.