InfiniTimeOrg/InfiniTime #1246: Heart Rate UI Update Concept

RageGamerBoi, 2022-07-25:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Replace the heart rate start button with the new timer UI button.

### Description

Update the heart rate app from this:

[image]

To something like this:

[image]

mashuptwice, 2022-07-25:
It would definitely be more consistent with the timers button design, but not with the stepcounter. 

I'm not sure if a bigger button makes sense, because it would be easier to accidentally disable the HRM when keeping the app open.

Apart from that IMHO it would look nicer.

ITCactus, 2022-07-28:
there are ["UI design guidelines"](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/ui_guidelines.md),
and according to that, the "start" button should be full width indeed

RageGamerBoi, 2022-07-28:
> * Make interactable objects big

and 

> * Align objects all the way to the edge or corner

Should cover the wide buttons.

As for the other question, the Start/Stop button already enables/disables the HRM through the entire OS, none of the functionality needs to be different here.

RageGamerBoi, 2022-07-28:
As for the stepcounter concern;

> It would definitely be more consistent with the timers button design, but not with the stepcounter.

Check issue #1254 that I just made :)

Idcrafter, 2023-08-25:
also would a moving graph make it look nicer while also serving as visual feedback for it being active