InfiniTimeOrg/InfiniTime #1247: Add uptime since last battery charge to the battery setting

devnoname120, 2022-07-25:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

fsadfas

Zandengoff, 2022-07-28:
Was going to put in this feature request and saw that had already been done. Would love this on the main battery screen.

mashuptwice, 2022-08-22:
As soon as the external flash is functional, it could be pretty simple:

For SOT:
- counting the time between a GoToRunning and GoToSleep Message - add it to a altogether value in flash

Could be done in SystemTask
! It could cause a high write load on the flash.

For uptime since charge:
- write the uptime on charger disconnect to flash
- uptimeSinceCharge = ( uptimeNow - uptimeChargerDiscon )

Could be done in BatteryController

But it depends on the flash unfortunately, as a lot of other things.