InfiniTimeOrg/InfiniTime #1248: Snooze for the alarm clock

calhix, 2022-07-25:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

A snooze feature would be very useful for the alarm clock

### Description

I'm really missing a snooze feature but I'm not sure if it could be added to the alarm clock app itself. Perhaps of instead going to the alarm clock screen when an alarm goes off, it would show a different screen that looks similar to the alarm clock app but just shows the current time, a large stop button and a large snooze button that can be configurable in the settings. The alarm would then ring again when ever the amount of minutes has lapsed that you previously configured with the same screen so you have the option to snooze as many times as you want.

Possible icon: https://fontawesome.com/icons/bed?s=solid

electricsquiral, 2023-06-16:
Is it too late to ask for an off switch on this feature?

KiralyCraft, 2024-07-16:
I support this as well, although the current method implies manually increasing the minutes while the alarm is going, and then turning it off.