InfiniTimeOrg/InfiniTime #1249: swipe down to close quick settings

mashuptwice, 2022-07-26:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

To make navigation more consistent across the entire UI, exit the quicksettings by swiping down. 

### Description

Currently it is possible to get back to the watchface from every app by simply swiping down. 
The only exception to this is the quick settings, where you have to swipe to the left to exit them.

I sometimes find myself trying to exit the quicksettings by first swiping down, resulting in a random option getting selected based on the starting position of my finger. 

To have a consistent input schema across all of the screens, I propose to also close the quick settings on swiping down.

Riksu9000, 2022-07-26:
This wouldn't be consistent considering the positions of the screens. To move left you swipe right and to move right you swipe left. The apps raise from the bottom, so you close them by swiping them down. Notifications is also different, because it is above the watchface.

mashuptwice, 2022-07-26:
@Riksu9000 I understand that. Note that I only propose to add a second gesture, not to change the main gesture. 

Currently on android you can both close an app using the home as well as the back gesture, even if the animation seems to open the app from and close it to the bottom of the screen and it is quite convenient that way. 

On infinitime if I would want to exit the settings app, lets say from the bluetooth settings you now have to:
1. swipe down four times
1.1 swipe down a fifth time
1.2 realize you've swiped one time too much and accidentally opened the flashlight
1.3 swipe down another time
1.4 reposition the finger 
2. swipe left to exit the quick settings

Instead of simply swiping 5 times. 

While the current behavior would keep the animation consistent with the animation/position of the screen, it clearly lacks consistency with the inputs. Considering that the main function of a watch is displaying the time, in my opinion it should be as easy as possible to return to the watchface.

It might be worth a discussion if the focus is on consistent design rather than usability.

Riksu9000, 2022-07-26:
There is in fact a way to quickly return to the watchface, which is by long pressing on the button on any screen. I hope this solves your issue as I would rather not add an extra gesture there.

mashuptwice, 2022-07-26:
I am aware of that but thanks for pointing that out.

mashuptwice, 2022-08-28:
Implemented in https://github.com/mashuptwice/infiniTime

closed 2022-08-28 by mashuptwice