InfiniTimeOrg/InfiniTime #1250: Clock settings are lost after battery has been drained out

cquike, 2022-07-27:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Clock settings are lost after battery has been drained out

### What should happen instead?

Clock settings are remembered

### Reproduction steps

The battery was drained out. After a successful recharge the clock settings weren't preserved

### More details?

The following settings (at least that I realized) were lost after the battery was recharged:
- Time
-  Date (it shows again some date in 1970, like it come from factory).
- Watch face is now analog, while before it was set to digital.

Resetting the time is easy using to the gardgetbridge application again. I am not sure whether other settings are lost since for the time being I have not used much the watch and I don't remember all settings that I changed.

### Version

1.6.0

### Companion app

Gadgetbridge

Riksu9000, 2022-07-27:
There's no way to keep track of the time if the battery's empty. The watchface changing is strange, especially since it should default to digital. Are you sure you didn't change it before? Also it's recommended to update your version of InfiniTime as 1.6.0 is quite old and issues that happen on old versions may not be worth investigating since they may already be fixed.

Danimations, 2022-08-01:
Yes, this problem is real, but it's not one that can be solved by the firmware developers. Remember when your old non-smartwatch used to run out of battery? Same problem. Time can't be held without a functioning battery. 

I also agree with Riksu9000's suggestion about updating your firmware. Gadgetbridge can do this for you. You just download the firmware file, use "open with" in Android's file manager, and select Gadgetbridge. Gadgetbridge recognises it as a firmware file, and will prompt you to flash it. I've done this a few times now. Firmware version 1.10.0 is excellent- a vast improvement on 1.6.0. You'll find that when you pair your smartwatch with your device running Gadgetbridge, the time and date will be syncronised automatically.

ght, 2022-09-15:
This issue should either be closed as wontfix, or be given actionable goals that show what needs to be implemented for this to be considered fixed. One option of improving the behavior when the battery runs out would be to regularly (how often?) save the time to flash, since most users will charge their watch relatively shortly after the battery runs out and it would be nice not to have to set the date in that case (or only bump the day up by one). I'm not sure how desirable the same behavior would be for the time; on the one hand it would make it easier to set the time when charging the watch right after it runs out (only have to change by a few minutes), on the other it could mislead the user into thinking the time was set when in reality it is at least a few minutes late.

A simple implementation, which requires little changes and uses existing functionality, would be to save the time every five minutes or so (could actually be limited to only happen when battery is low). When booting and a stored time is restored, push a notification telling the user that the time was restored, but might be off, and should be set either by companion app or manually.

JF002, 2022-09-17:
I think @ght is right and we can close this bug report : the watch cannot keep track of the time when the battery is empty, and I guess the issue with the watchface is probably fixed in newer versions of InfiniTime.

Regarding @ght 's suggestion : saving the time regularly in flash is not a good idea : flash memory support a limited number of write/clear cycles, and writing to the memory every 5 minutes could wear the memory faster than expected.

closed 2022-09-17 by JF002

ght, 2022-09-17:
> Regarding @ght 's suggestion : saving the time regularly in flash is not a good idea : flash memory support a limited number of write/clear cycles, and writing to the memory every 5 minutes could wear the memory faster than expected.

You are right about flash wearing out, but according to the (Wikipedia page about flash memory)[https://en.wikipedia.org/wiki/Flash_memory#Write_endurance], NOR flash is very high endurance with expected erase cycles upwards of 100K. That means when we store the current time as 64bit number (4B), write it every 5 minutes (12/hour, 288/day), that is ~1.2KiB/day, or >9years to fill the flash once (so >900k years minimum to wear out the flash, if we disregard (write amplification)[https://en.wikipedia.org/wiki/Write_amplification]). Considering that we only need to save the time when the battery is about to run out (e.g. <10%, when the fixed discharge curve is merged), I don't think flash wear is a valid argument in this specific case, even when taking write amplification into account :)