InfiniTimeOrg/InfiniTime #1255: Activity/Workout Tracking (different modes, on device

RageGamerBoi, 2022-07-28:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Integrate different workout tracking modes and Gadgetbridge-like jog/run/sprint tracking

### Description

Gadgetbridge has passive walk/jog/run tracking by simply taking how many steps happen over a time period and determining if you're running or jogging or just walking. This would be great to have integrated onto the watch, as it currently relies too heavily on the real-time step data sent over BLE which is flaky at best (and if you don't have your phone on you it gets even worse.).

This could be combined with different manual workout categories like a lot of other watches on the market, and synced to Gadgetbridge (I think they already have a way to do it here: https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Sports-Activities-Workouts but I'm not quite sure).

Some workout-specific tracking options to include in a dedicated app could be biking, weight lifting, and manual switching to the jog and run modes in case the passive tracking fails. I read in a different Issue that the pedometer uses its built in algorithm for step tracking, so accelerometer use might be necessary.

minacode, 2022-08-01:
The motion sensor can differ between some modes. Still/walk/run or so.

dozadoesit, 2022-08-25:
I have been waiting for a feature like this as well. Gadgetbridge can definitely do this with other devices.

RageGamerBoi, 2022-09-29:
Maybe once external storage is working we could store a database of timestamped steps and later send it over to a companion device.
This then could be used to get workout times and be parsed more thoroughly then just a simple "User ran from 11:50 to 12:21" type message.

minacode, 2022-09-30:
#749 does elaborate the same ideas. I think the main issue is that someone needs to actually do this 😄