InfiniTimeOrg/InfiniTime #1260: Watchdog reset after sometime when connected to Amazfish running on PinePhone

AndreySV, 2022-08-01:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Device reboots and after reboot 'last reset' is 'wtdg'

### What should happen instead?

Device doesn't reboot that often

### Reproduction steps

1. Pair PinePhone with Mobian running kernel 5.15.44 and Amazfish 2.0.3 with Pinetime
2. Normally use Pinetime (notifications, step counters)
3. Notice, that device reboots with 'last reset' reason 'wtdg'. Pinetime usually reboots after 6-10 hours of work. 

If device is not paired with a phone, it doesn't reboot in days ( >13 days).

### Version

1.10

### Companion app

Amazfish 2.0.3

Itai-Nelken, 2022-08-02:
I have a similar problem (resets caused by watchdog but at random times. Sometimes once per day, and sometimes after a whole week) with the pinetime paired to gadgetbridge, but I thought it was caused by something I did because I'm using a custom build.

JF002, 2022-08-02:
Watchdog reset can have many sources : the software crashed (invalid memory access, infinite loop,,...), a bug in a library (ble stack, UI), a hardware issue with a device (accelerometer, display, memory),... so this issue won't be easy to diagnose, unfortunately!
I haven't noticed any crash recently on my PineTime connected most of the time to Gadgetbridge.

Do anyone else experience that kind of issue? Do you have any other info that could help us determine the cause of the crash?

AndreySV, 2022-08-14:
I've seen the problem with 1.9 and probably with 1.8 in the past.
I can reproduce reset by watchdog by flooding PineTime with notifications. See following video: https://www.youtube.com/watch?v=APFLuV7WYLc