InfiniTimeOrg/InfiniTime #1263: please add color correction for people with Colour vision deficiency

lman0, 2022-08-02:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

add list of color coorection by type of color blindness so these people can use pinetime

### Description

some people are born with Colour vision deficiency 
it make the  unable to see some color (like blue , green red,)
it would be great if it was possible to add in setting a color correction menu that list the kind of color blindness
'none' , 'red-green'  ...
and for each color correction add layer to color so the final color became visbile to them .
like adding colors to #1258 

it not a theme by the way, since the original color are not modified (in none case) but only corrected for them

Riksu9000, 2022-08-03:
Accessibility is definitely something important to look into. This issue can also be solved by carefully designing the UI in a way which makes it understandable without even considering the colors. Do you know if some UI elements are especially problematic?

I took a look at InfiniTime with a color blindness simulator and found a few issues which I'll look into fixing. Otherwise I think InfiniTime is already quite good about this. Most objects have identifying elements in white, which is the easiest to see for everyone.

ITCactus, 2022-08-03:
it also could be a part of the Global Color Scheme #1030
i mean like in some IDEs

lman0, 2022-08-04:
@Riksu9000 color blindness make people see without the color 
and if color hold a meaning , then these people are unable the see such meaning.

explanation  (please see the website , since it show visually how each kind see the color)
https://www.colourblindawareness.org/colour-blindness/types-of-colour-blindness/

here some kind :
* Red–green color blindness :  mediacal name "deuteranopia" *
* blue–yellow color blindness :  mediacal name "Tritanomaly" *

lman0, 2022-08-09:
i found a color listing that allow to change color red to color red for deutonmaly ...
https://www.disabled-world.com/disability/accessibility/websitedesign/color-spectrum-chart.php

please create a select list that change base color to each kind like "deuteranopia"

lman0, 2022-08-09:
what is the purpose of the green color of the notification icon?
if it has a meanning  please change the icon if you plan to remove the color

Riksu9000, 2023-02-20:
To avoid having to add big features not used by the majority, let's apply [universal design](https://en.wikipedia.org/wiki/Universal_design) concepts to the UI design. I'll close this issue for now. Anyone, please open a new issue if you have trouble using InfiniTime for any reason and we'll see what we can do about it.

closed 2023-02-20 by Riksu9000