InfiniTimeOrg/InfiniTime #1275: Failed to build InfiniTime on Docker Container under Windows 10

A4EVA, 2022-08-11:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Docker container do not build the firmware under Windows 10

### What should happen instead?

The Build process should have been completed

### Reproduction steps

Just clone and initialize the repository, after that pull the docker image with `docker pull infinitime/infinitime-build` considering that we are under Windows to avoid problems instead of using `docker run --rm -it -v $(pwd):/sources infinitime/infinitime-build` use `docker run --rm -it -v C:\...FULL-PATH\Projects\InfiniTime:/sources infinitime/infinitime-build` the process will start but once reached 36 % of the build this error will come up : 

`[ 36%] Linking C static library libnimble.a
[ 36%] Built target nimble
gmake: *** [Makefile:91: all] Error 2
+ return 1
+ BUILD_RESULT=1
+ '[' '' '!=' true -a 1 == 0 ']'`

once inside the container with: 
`docker run --rm -it -v C:\...FULL-PATH\Projects\InfiniTime:/sources infinitime/infinitime-build bin/bash` 
I have launched the build process manually with make and a bit more exact error can now be seen:

`[ 36%] Generating jetbrains_mono_bold_20.c
(Stripping trailing CRs from patch; use --binary to disable.)
patching file jetbrains_mono_bold_20.c
patching file jetbrains_mono_bold_20.c
patch: **** malformed patch at line 3:

Traceback (most recent call last):
  File "/sources/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/sources/src/displayapp/fonts/generate.py", line 70, in main
    subprocess.check_call(['/usr/bin/env', 'patch', name+'.c', patch])
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/env', 'patch', 'jetbrains_mono_bold_20.c', 'jetbrains_mono_bold_20.c_M.patch']' returned non-zero exit status 2.
make[2]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_bold_20.dir/build.make:73: src/displayapp/fonts/jetbrains_mono_bold_20.c] Error 1
make[2]: *** Deleting file 'src/displayapp/fonts/jetbrains_mono_bold_20.c'
make[1]: *** [CMakeFiles/Makefile2:590: src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_bold_20.dir/all] Error 2
make: *** [Makefile:91: all] Error 2`

I have tested on two different Windows 10 machines and the error is exactly the same. I have not understood what I did wrong

### Version

eb487c7

A4EVA, 2022-08-12:
Thanks a lot, Everything worked nicely. I would like to use this container to build the code with Visual studio Code it is possible to do that ? I never used VS Code with a container so I have no idea :-)

A4EVA, 2022-08-12:
Ok If I reopen the project with the container as VSCode suggest it will pull another image (vsc-infinitime) and is not the one I tried (the latest infinitime-build) I also do not know what is project configuration you mentioned. I tried to build the project once VSCode finished with the docker configuration and this error pops up: `Could not find LV_FONT_CONV using the following names: lv_font_conv` maybe is caused by the missing LV_FONT_CONV in the docker image that VSCode downloaded by itself.

FintasticMan, 2022-08-12:
The devcontainer is quite outdated, no-one has gotten around to finishing updating it yet.

A4EVA, 2022-08-12:
@ITCactus I installed npm and then lv_font_conv but the cmake output still say that lv_font_conv is missing. Do I need to do something else after the manual installation of lv_font_conv ?

A4EVA, 2022-08-12:
Ok, found the problem I almost finished to setup everything on the old docker container. I have one last problem to figure out at the end of the build (100%) this error pops up `[build] /opt/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/pinetime-mcuboot-app.dir/displayapp/screens/StopWatch.cpp.o: in function `Pinetime::Applications::Screens::StopWatch::stopLapBtnEventHandler(unsigned char)':
[build] /workspaces/InfiniTime/src/displayapp/screens/StopWatch.cpp:156: undefined reference to `Pinetime::Applications::Screens::StopWatch::maxLapCount'
[build] collect2: error: ld returned 1 exit status
[build] gmake[3]: *** [src/CMakeFiles/pinetime-mcuboot-app.dir/build.make:1863: src/pinetime-mcuboot-app-1.10.0.out] Error 1
[build] gmake[2]: *** [CMakeFiles/Makefile2:310: src/CMakeFiles/pinetime-mcuboot-app.dir/all] Error 2
[build] gmake[1]: *** [CMakeFiles/Makefile2:317: src/CMakeFiles/pinetime-mcuboot-app.dir/rule] Error 2
[build] gmake: *** [Makefile:202: pinetime-mcuboot-app] Error 2
[build] Build finished with exit code 2`

I never encountered problems with this piece of source by the way all the EOLs has been corrected in both mcuboot and the fonts patches @ITCactus mentioned before

iksi4prs, 2022-08-29:
Hi, I'm also trying to build on Window 10.
I've fixed the issues as mentioned earlier for font patch files.

Now I get this error:
```
post build steps for pinetime-mcuboot-recovery-1.10.0
   text    data     bss     dec     hex filename
 184372     504   34436  219312   358b0 pinetime-mcuboot-recovery-1.10.0.out
/usr/bin/env: ‘python3\r’: No such file or directory
gmake[2]: *** [src/CMakeFiles/pinetime-mcuboot-recovery.dir/build.make:1035: src/pinetime-mcuboot-recovery-1.10.0.out] Error 127
```

because of "\r" in error, I think it is another issue of end-of-line.
can anyone suggest which file to look for ?
I've used latest code (30/Aug) of branch "develop".
10x.

iksi4prs, 2022-08-30:
Ive managed to fix it by also fixing py files, as described in https://github.com/InfiniTimeOrg/InfiniTime/issues/1056#issuecomment-1080476567

In any reason why both fixes in font files and py files are not fixed/committed in code ?

Avamander, 2022-10-04:
Use of the container on Windows is not officially supported and if someone wishes they can contribute documentation on how to get it working. I see no reason to keep this issue open for that.

closed 2022-10-04 by Avamander