InfiniTimeOrg/InfiniTime #1277: Forecastie implementation.

Sokoloft, 2022-08-12:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Forecastie weather stats. 

### Description

Hello,

I'd like the ability to see the outside temperature on my watch.
Probably up in the notification area of the Pinetime. Near battery and bluetooth icons.

I use an app called [Forecastie](https://github.com/martykan/forecastie) for my weather information.
Relevant [post](https://github.com/martykan/forecastie/issues/681) on the Forecastie github.

Thanks.

FintasticMan, 2022-08-12:
There is a weather implementation in InfiniTime, but it is yet to be used anywhere. Also, weather support would need to be provided by the companion app.

minacode, 2022-08-13:
Is it really necessary that all BLE services are provided by the same app when BLE connects devices? Just asking because I am curious.

Avamander, 2022-08-13:
@minacode if multiple connections were possible, then it wouldn't be necessary. Right now it is.

minacode, 2022-08-13:
So we need a separate connection per app, even with the same device?

Avamander, 2022-08-13:
Yes.

Even if it were possible, it's not the best design. It's better to let the weather app send weather intents and the companion app deal with the complexity of pushing it to the device.

Sokoloft, 2022-08-13:
So it seems this is mostly possible. @joserebelo on the Gadgetbridge codeberg page had said currently gadgetbridge does not forward any weather data to the Pinetime currently. They also pointed out that there is a merge request for Forecastie that forwards data to gadgetbridge. I have built this merge, however I'm awaiting support for gadgetbridge to send weather data to the Pinetime/Infinitime.

Sokoloft, 2022-08-14:
Hello,

So if I'm understanding everything correctly. There is support for weather currently with Infinitime, however the only UI for it is meant for debugging. Supposedly you can modify the firmware to have that weather debugging UI app in the app list. Preferably, at least myself. It'd be nice to have relevant weather information on the main watch face. Or maybe swiping to the left on the main watch face would work? 

I really hope this can be implemented in the next few updates. I have very little to no coding experience else I'd give it a shot myself. 

I was recommended to try wasp-os but I don't like its look compared to Infinitime.

JF002, 2022-08-15:
You are right : InfiniTime already implements a [weather service](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/components/ble/weather/WeatherService.h) and there's a [debug app](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/displayapp/screens/Weather.h) in the source code (but not built into the release files).

[ITD](https://gitea.arsenm.dev/Arsen6331/itd) already supports the weather API, so I think we (InfiniTime devs and contributors) just need some time to work on that feature. IIRC, we ran into [some issues](https://github.com/InfiniTimeOrg/InfiniTime/issues/1049) and haven't had the opportunity to do some progress since then.

It'll be easier for companion apps developer to add support for the weather API when we have a fully working weather app.

L3br4nd, 2024-12-08:
1.15 features current weather information and a forecast app. 
Is this still open because forecastie is not on the list of supported weather providers?

FintasticMan, 2024-12-08:
It's just open because we haven't closed it yet :-) Support for sending weather to Gadgetbridge needs to be implemented in Forecastie, which doesn't seem to be maintained at the moment. There is a pull request on their repo that implements it, so if anyone wants it, they can build that PR. I'll close this issue now.

closed 2024-12-08 by FintasticMan