InfiniTimeOrg/InfiniTime #1290: pinetime-graphics not present anymore in CMakeLists.txt

A4EVA, 2022-08-17:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

pinetime-graphics has disappered from the build options

### What should happen instead?

Should be present ... I guess

### Reproduction steps

Just clone the repo the option to build the pinetime-graphics has been removed even if the readme in the bootloader folder said to make it with make pinetime-graphics and the files to change the logo are present

### More details?

I was about to change the boot logo with another one I designed myself. After searching for a solution to better understnad how to change the logo I found the readme in the bootloader folder, in addition there is the boot_graphics.h file and the documentation said basically to call the make pinetime-graphics in order to have pinetime-graphics binary made: I was unable to do that simply because in the CMakeLists.txt in the src folder there is no mention anymore of the pinetime-graphics option. I do not know if is a documentation error or what but please if someone knows how to change the logo with the current version of the firmware please let me know :-) Thanks a lot to everyone will join

### Version

v1.10.0

JF002, 2022-08-21:
We removed the target `pinetime-graphics` because we changed the way the graphic is handled between the first version of the bootloader (by @lupyuen) and the current one.

The [older bootloader](https://github.com/lupyuen/pinetime-rust-mynewt) fetched the picture from the external SPI flash. It was then possible to write a custom picture to the memory using the firmware `pinetime-graphics`.

The [newer bootloader](https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader) (the one that is flashed by default on the PineTimes) has the picture embedded in the binary file. There were various reasons for this change : we wanted to use that space in spi flash to store the recovery firmware, it made the code simpler,.. The downside is that you have to rebuild the whole bootloader firmware to change the picture.

I'm sorry for the confusion : the documentation about the bootloader is partially out of date. I'm currently working on it in #1288, and I've already removed mentions to `pinetime-graphics`.

In the current bootloader, the graphics (in binary) is located in https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader/blob/master/libs/pinetime_boot/src/graphic.h. IIRC, it only supports 1-bit pictures (2 colors) and it must be RLE encoded by [this tool](https://github.com/InfiniTimeOrg/pinetime-mcuboot-bootloader/blob/master/tools/rle_encode.py).
Be cautious if you test your own build of the bootloader on your PineTime : there's no recovery from a failed bootloader (except by reflashing the bootloader and firmware using SWD, which is not possible on sealed devices).

A4EVA, 2022-08-23:
Hello JF,

thanks a lot for pointing that out, now everything is clear ! I was trying to build the new bootloader after setting up all the mandatory pre-requiments but when I init the repo with : `git submodule update --init --recursive` I get this error:

`Cloning into 'C:/Users/AmericoX86/Documents/PROJECTS/pinetime-mcuboot-bootloader/reloader'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:daniel-thompson/wasp-reloader.git' into submodule path 'C:/Users/User1/Documents/PROJECTS/pinetime-mcuboot-bootloader/reloader' failed
Failed to clone 'reloader'. Retry scheduled
Cloning into 'C:/Users/User1/Documents/PROJECTS/pinetime-mcuboot-bootloader/reloader'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:daniel-thompson/wasp-reloader.git' into submodule path 'C:/Users/User1/Documents/PROJECTS/pinetime-mcuboot-bootloader/reloader' failed`

It will be better to open an issue on the bootloader repo right ?

JF002, 2022-08-28:
It looks like an authentication issue with Github instead of an issue with any of the repo.

I've just checked on my setup : `git submodule update --init --recursive`  works fine. 

The file *.gitmodule* that specifies the submodules contains

```
[submodule "reloader"]
        path = reloader
        url = git@github.com:daniel-thompson/wasp-reloader.git
```

Notice that it uses the SSH path to git. You probably need to configure a SSH key in your github account to be able to checkout the repo.

However, you can try to edit that file with this content:

```
[submodule "reloader"]
        path = reloader
        url = https://github.com/daniel-thompson/wasp-reloader.git
```

Now, the checkout will be done using the HTTPS path, which does not need any authentication for read operations.

Let me know if it works. If it does, I'll edit the file in the repo.

A4EVA, 2022-08-29:
Hi,
the modification you suggested works perfectly but, something strange is happening if I try to build:

`jokerx86@A4A400:~/pinetime-mcuboot-bootloader $ ./scripts/nrf52/build-boot.sh
+ source paths.bash
+ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

+ newt build nrf52_boot
Building target targets/nrf52_boot
Target successfully built: targets/nrf52_boot
+ newt size -v nrf52_boot
Size of Application Image: app
Mem FLASH: 0x0-0x7000
Mem RAM: 0x20000000-0x20010000
  FLASH     RAM 
Loading compiler /home/jokerx86/pinetime-mcuboot-bootloader/repos/apache-mynewt-core/compiler/arm-none-eabi-m4, buildProfile optimized
objsize
   text	   data	    bss	    dec	    hex	filename
      0	      0	      0	      0	      0	/home/jokerx86/pinetime-mcuboot-bootloader/bin/targets/nrf52_boot/app/@mcuboot/boot/mynewt/mynewt.elf
+ arm-none-eabi-objcopy -I binary -O ihex bin/targets/nrf52_boot/app/@mcuboot/boot/mynewt/mynewt.elf.bin bin/targets/nrf52_boot/app/@mcuboot/boot/mynewt/mynewt.elf.hex
arm-none-eabi-objcopy: error: the input file 'bin/targets/nrf52_boot/app/@mcuboot/boot/mynewt/mynewt.elf.bin' is empty
`

did you have encountered this problem ? Is like newt is not producing the bin file ... I guess