InfiniTimeOrg/InfiniTime #1298: Watch does not reconnect after phone reboot

Foxtrek64, 2022-08-26:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

PineTime/InfiniTime disconnects and does not reconnect.

### What should happen instead?

Reconnecting through GadgetBridge should allow connecting to the device.

### Reproduction steps

This issue seems related to #302 which has already been closed as patched, but it has come back up for me.

In my case, I have not tested with Airplane mode, only with rebooting my phone.

1. Connect PineTime to phone with GadgetBridge.
2. While phone is connected, reboot the phone.
3. Phone reboots, Gadgetbridge does not reconnect.
4. Open GadgetBridge and tap Infinitime entry.
5. Wait for connection
6. "Connecting..." status at bottom disappears. Saved gadget goes back to "not connected."

### More details?

After updating to 1.10.0, forgetting the device and reconnecting appears to resolve the issue.
In 1.6.0, I had to hold down the power button until the pine button became red, then reboot the device as the device would not reconnect.

### Version

1.10.0 (286690e8)

### Companion app

Gadgetbridge 0.68.0

Avamander, 2022-08-26:
1.6 is really not relevant any more. Does this occur with 1.10.0?

Foxtrek64, 2022-08-26:
Yes, sorry for any confusion.

The issue started with 1.6.0 (the version I had when my watch arrived). There only doing a hard reset as described would fix the issue.

After updating to 1.10.0, the issue persists, however now forgetting and rejoining the device seems to resolve the issue. It's more an annoyance than anything, seeing as my phone gets powered off on a semi-regular basis.

Avamander, 2022-08-26:
Why are you re-pairing the device, why not just force a reconnect by either long-pressing the card in GB or toggling Bluetooth?

mxnemu, 2022-08-27:
I guess you have to re-pair gadgetbridge after the 1.10 update, because from 1.6 to 1.10 authentication was added.

Also make sure to mark the firmware as validated, so the watch won't reboot with the old 1.6 firmware. I recently got a pinetime and was a bit confused at the start, but everything works fine for me on 1.10.

Foxtrek64, 2022-08-27:
In recent days, GB has been performing better. In the past, the force reconnect or bluetooth toggle did nothing. Now whenever reconnecting to the watch, it just works:tm:.

I am unsure what it was exactly that changed, though I did notice the otp-like authentication after doing a force reconnect on 1.10. After my first time authenticating things have been working great. I even updated the GB app today and was able to reconnect without issue.

While I'd like a root cause, just having it work is good enough for me. I'll go ahead and mark this as closed.

Thanks!

closed 2022-08-27 by Foxtrek64