InfiniTimeOrg/InfiniTime #1301: Documentation for tips on adding new watch face

bmsleight, 2022-08-28:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Add Documentation for tips on adding new watch face

### Description

PR to follow 
Useful way to start hacking a new watch face.

kieranc, 2022-09-06:
There's a wiki page here: https://wiki.pine64.org/wiki/PineTime_Custom_Watchface_Tutorial
It might be somewhat out of date, I'll try and scan through it to see, and maybe add it to the new docs repo.

margamanterola, 2024-07-07:
I recently went through the process of adding a new face. All documentation online is currently so outdated, that it might be better to not have it at all, as it's super misleading.