InfiniTimeOrg/InfiniTime #1305: New WatchFace to give description of current time accurate to a minute

bmsleight, 2022-08-28:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

New Words WatchFace - PR to follow

### Description

PR to follow see screen shots

bmsleight, 2022-08-28:
[image]
[image]
[image]

hrmckay, 2022-08-28:
There is a similar watch face called Wordclock on this site:  https://zephyrlabs.github.io/Watchfaces/

bmsleight, 2022-08-28:
Yes similar, but my preference  "Almost quarter past one in the afternoon"  to "fourteen past one" [13:14] and I prefer "Just gone twenty to nine in the evening" to "nineteen minutes to nine" [20:41]

bmsleight, 2022-08-28:
Dates in words as well
[image]

kieranc, 2022-09-06:
There's also this PR: #660

Riksu9000, 2023-02-20:
Not sure what the point of this issue is. We don't have a convention of having an issue for each PR.