InfiniTimeOrg/InfiniTime #1308: Disable Step Counter/Goal

Fidgetcetera, 2022-08-30:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I'd like to be able to disable the step counter or its goal.

### Description

Simply setting the goal count to 0 and never getting notified about it would be fine. I'm just disabled, in a wheelchair most of the time, and it's not a useful metric for me the way heart rate monitoring or getting notifications on my watch are, and it's really not a feature I bought a PineTime for and I'd like to opt out.

tgc12, 2022-09-01:
Goal and it's notifications are dependent of the companion app (Gadgetbridge...)

Fidgetcetera, 2022-09-01:
> Goal and it's notifications are dependent of the companion app (Gadgetbridge...).
> 
> Disabling step recording is a good idea, to at least have the posibility.

Ah, I somehow completely forgot this and got mixed up. That said, I'll leave this open for disabling step recording and modify the title.

Joheyy, 2022-09-23:
I think this is something a lot of people would appreciate so I mad a pull request.

Joheyy, 2022-09-24:
For anyone interested in this I made a small build incorporating my fix into 1.10.0
(Note that I will probably not update this in the future unless there's enough interest in it)

https://github.com/Joheyy/InfiniTime/releases/tag/1.10.0-disable-step-count

@Fidgetcetera

shade-belisar, 2024-02-14:
I would also be interested in this feature. I'm somewhat confused about the state of this issue though. Am I correct in assuming that @taukakao implemented this, but it was never incorporated into the main repository?

taukakao, 2024-02-15:
@shade-belisar 

Yeah, from what I can tell they don't want to add settings that are not absolutely needed. 
I can understand that sentiment though I still think this setting would've been worth it.

I would mark this issue as wontfix though honestly.