InfiniTimeOrg/InfiniTime #1311: Bug: `motorController.RunForDuration(0)` vibrates indefinitely

ght, 2022-09-02:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Calling `motorController.RunForDuration(0)` does not stop vibrating

### What should happen instead?

Vibrating for "0" duration should not vibrate at all

### Reproduction steps

Call `motorController.RunForDuration(0)`.

### Version

v1.10.0, also on current master (`f53e7506`)

minacode, 2022-09-02:
Just wondering, but did you test this for small durations? Would be interesting if it always works as expected for 1.

ght, 2022-09-02:
Calling `motorController.RunForDuration(1)` does indeed work as expected, producing a light "click" which can barely be felt, but heard in quiet environments. It makes quite nice haptic feedback for button presses, which is what I tried out using it for in my personal fork.

It does indeed always work, just as passing `0` always causes the motor to buzz until the next buzz stops it; you can make it stop by starting the metronome for example. But that's just a workaround for devs, nothing an end-user should have to figure out.

The current code does not trigger this bug, but I see no reason for not fixing it.

PR #1312 (2022-10-02): motorController: Fix infinite vibration on RunForDuration(0)
Fixes #1311.
closed 2022-10-02 by JF002