InfiniTimeOrg/InfiniTime #1314: Motor vibrator burn in button

ky56, 2022-09-03:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

A button that runs the motor full speed for a couple minutes

### Description

My PineTime has a faulty vibrator motor that intermittently won't vibrate. I believe that the bushes are faulty and running the motor full speed could "burn them in" and fix problem. Maybe add this under a debug menu or the existing about menu.

JF002, 2022-09-04:
Note that this feature would probably be used very rarely, as its only purpose is to debug a very specific hardware issue.

Regarding you issue with the motor : in the past, a few units had their motor blocked by the BLE antenna that was too long and that was interfering with the motor. This QC issue should have been fixed since then, but if it's your case, I'm not sure that a burn-in feature would help to fix it, unfortunately.

ky56, 2022-09-04:
I am able to make it vibrate again temporarily with a strong flick of the wrist. So I doubt it's that. I was using the metronome to test this. My thinking is that there is a dead zone on the bush or armature and running it for a few minutes to half an hour should wear it in.

This is just a theory of course. In the meantime I'm waiting for a reply from Pine64 support regarding this. I was hoping to test this theory before then.

joseph58tech, 2022-09-06:
just run the metronome at 400 - 500 bpm and you will get the same effect

TheAwesome98-Real, 2022-10-01:
the metronome doesn't go over 220

Riksu9000, 2022-12-30:
You can flash a special debug firmware to do this experiment. We can't really afford the memory to add niche features like this.

closed 2022-12-30 by Riksu9000