InfiniTimeOrg/InfiniTime #1319: Reboot option

UNIVPM-S1079707, 2022-09-08:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Add an option in the settings to reboot the device

### Description

Currently, the only way to reboot the device if something wrong happens are to wait that the battery runs out and updating the firmware.

Since it reboots to update the firmware, I don't think it should be hard to have a button in the settings that makes it reboot without updating the firmware.

SteveAmor, 2022-09-09:
Does holding the physical button to force a reboot fit your requirement?

UNIVPM-S1079707, 2022-09-09:
It technically does.

JF002, 2022-09-11:
As @SteveAmor said, you can use the physical button to reboot the firmware. Adding a new entry in the settings is probably not needed.

closed 2022-09-11 by JF002