InfiniTimeOrg/InfiniTime #1321: Long-Term Battery Health Issues

RageGamerBoi, 2022-09-10:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Battery drains to 0% and the watch doesn't appear to "shut down" to prevent high depth of discharge that is detremental to the health of the battery.

### What should happen instead?

A common behavior from electronic devices is to gracefully power off when the device reaches 0%

### Reproduction steps

Discharge the battery almost all the way until the indicator reads 0% and the watch continues to work until the battery is unable to supply enough voltage and the watch dies.
This is generally a bad thing for long term battery health:
https://en.wikipedia.org/wiki/Depth_of_discharge

### More details?

This might not be as big of a problem as it seems, and may have already been compensated for on a hardware level (or on a software level that I do not know about as of yet), but it still seems like a better idea to gracefully turn the watch off when the battery hits 0% to prevent further discharge and degradation to the battery.

### Version

v1.10.0, bootloader 1.0.0

JF002, 2022-09-11:
When the battery is nearly empty, it cannot provide enough power to the MCU, which triggers the brown out detector and reset the MCU. The battery is protected with its own protection chip.

This is mostly a hardware issue : the hardware does not provide any way to shutdown the MCU. There's no physical button to cut the power, and there is no way for the software to disable the power supply. The only thing the software could do is to put the MCU in a halted state, but then, there would be no way to restart it again (none that I can find, anyway).

ght, 2022-09-15:
What about putting the watch into a state where it consumes as little power as possible when the battery voltage goes below some low water mark? This would decrease the danger of deep-discharging the watch and having it boot-loop when next put on a charger.

To give the users the choice, in case some prefer their watch to be functional a few minutes longer at the cost of risking deep-discharging it, this could have a toggle in the settings, like the aggressive power saver options on Android, but should be enabled by default.

We could at least:
- disable all sensors
- disable bluetooth
- prevent screen from being enabled
- prevent motor from being enabled
- maybe more that I'm missing? Can the SPI NOR flash be powered down? Can the MCU lower it's clock frequency?

In short: try to do nothing apart from resetting the watchdog timer every time it's about to run out.

RageGamerBoi, 2022-09-29:
@ght what if at around 5% battery it just shows the time and a low battery message under it (along with disabling all sensors, dimming screen, disabling Bluetooth, motor, etc.) to warn the user and preserve power

Idcrafter, 2022-10-06:
i noticed that the battery stays a long time on 0% till it eventually dies, can't there be like a shutdown on like 1% to not over drain it to increase it's overall lifespan?