InfiniTimeOrg/InfiniTime #1334: Develop: watch face depends on notification status

minacode, 2022-09-18:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

In recent test builds based on develop, the watch face changes not with its settin, bit the notification setting from the quick settings. 

### What should happen instead?

The watch face setting changes the watch face. 

### Reproduction steps

1. Build develop. 
2. Flash. 
3. Change the watch face and notification settings and see the watch face changing.

### More details?

In the setting data, the notification status and the clock face are stored adjacently. Maybe this causes a wrong lookup. The uint8 enum for notification values holds the same values that correspond to the clock face values. I could not find the place where this goes wrong yet. 

Maybe refactoring the clock face values into an enum (like the one for apps) gives a type hint somewhere.

### Version

develop

### Companion app

None.

JF002, 2022-09-18:
Could you be more specific on how to reproduce this issue? On my setup, a new watchface is applied when I edit the watchface setting, independently from the "silence" and "sleep" mode for the notifications.

minacode, 2022-09-18:
I updated the description. Could it be something with me testing #1294 some time ago? I will try to delete the settings file later and see if it helps.

minacode, 2022-09-24:
I solved it: I used the last version of the docker image to build the code. The bug is gone with the new version. It is interesting, that it failed that way.

closed 2022-09-24 by minacode

JF002, 2022-09-27:
This issue might have been caused by inconsistencies in [Settings.h](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/components/settings/Settings.h), especially if you built other branches that changed this file without clearing the old build artifacts. I'm not sure, but maybe files that include Settings.h were not properly rebuilt.

So yes, if you cannot reproduce this issue with an up to date Docker image and clean build environment, I guess we can close this issue.