InfiniTimeOrg/InfiniTime #1337: PineTime always reboots to Infinitime logo and is stuck there - Can't reflash firmware

AluminiumTech, 2022-09-20:
Hi,
I have a problem similar to the one mentioned in #576  and so I've used the nRF connect android app to try to reflash the firmware onto the sealed Pinetime, since that was recommended in the other issue, but as soon as the firmware upload gets to 4% the pinetime reboots itself.

Is there a way to get around this issue or flash the recovery another way?

Thanks.

FintasticMan, 2022-09-20:
The latest version of nRF Connect has broken DFU. You can either use an older version of nRF Connect, or use GadgetBridge, available on F-Droid, to flash the DFU. [This](https://github.com/NordicSemiconductor/Android-nRF-Connect/releases/tag/v4.24.3) should be the latest version of nRF Connect that works.

AluminiumTech, 2022-09-20:
Reflashing the recovery using GadgetBridge worked but reflashing Infinitime did not.

Reflashing Infitime initially was working but then the display of the pinetime started turning on and off randomly and eventually the upload failed.

Now the pinetime is boot looping. It loads the recovery, shows the Infitime logo and then a few seconds later reboots itself.

AluminiumTech, 2022-09-20:
After leaving it to charge for a few hours, I tried using GadgetBridge to reflash the firmware and it worked. Not sure why it worked this time but it worked.

closed 2022-09-20 by AluminiumTech

blaueente, 2022-10-18:
For future reference: This didn't work for me at first Gadgetbridge would not connect. nRF Connect would, but after starting the DFU, it almost immediately stopped again. The one crucial step that I had missed was the following: Unpair bluetooth in the system settings first.

greenbagels, 2023-09-24:
> For future reference: This didn't work for me at first Gadgetbridge would not connect. nRF Connect would, but after starting the DFU, it almost immediately stopped again. The one crucial step that I had missed was the following: Unpair bluetooth in the system settings first.

This was helpful; thanks! In case this doesn't work for anyone else, I also had to connect to the PineTime in Gadgetbridge by selecting the "Don't Pair" option.

djpeter84, 2024-01-07:
trying to update to 1.14.0, watch screen is InfiniTime but when i try to install, watch Infinitime turns green, upload hits 1% then i get Upload has failed.