InfiniTimeOrg/InfiniTime #1342: Notify on low battery

minacode, 2022-09-25:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Send a notification on low battery

### Description

The watch alerts the user once with a notification when the battery level goes under a threshold of maybe 50% (see #804) and the watch is not loading.

Companion apps can do this, but the watch should be able to do this by itself. I claim that no setting is needed.

I can do this in the next days, but if someone wants to do this instead, just comment below and I am fine with that.

RageGamerBoi, 2022-09-29:
This goes along perfectly with my issue #1321 
If somebody is working on this in the future, consider looking into that as well.

TheAwesome98-Real, 2022-10-01:
yes I've wanted this feature since I've had the watch

minacode, 2022-10-05:
If #1294 gets accepted, we could implement a low power mode that dims the screen, shortens the display time and maybe turns off bluetooth or touch wake up. The SystemTask could turn it on along with the issued notification.

minacode, 2022-10-11:
@RageGamerBoi: I completed the PR. The way I implemented it should also work for your shutdown feature in another PR. Just test for a threshold and send a message to the SystemTask which should be able to do what you wanted.

closed 2022-12-28 by minacode