InfiniTimeOrg/InfiniTime #1354: Automatic reconnect not working or delayed

tinxx, 2022-10-04:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

If connection to GadgetBridge is lost, the watch stays disconnected.

### What should happen instead?

The watch should automatically reconnect when back in phone's range.

### Reproduction steps

1. Securely pair PineTime with GadgetBridge
2. Place phone somewhere and walk away until Bluetooth disconnects
3. Walk back to phone and wait 

### More details?

I have been using an AmazFit Bip with Gadgetbridge before with the same phone and reconnection was quite reliable.
That is why I believe it is a watch-specific problem.

One can force GadgetBridge to reinitiate the connection either manually or by enabling "auto reconnect to device", which should not be necessary. 

### Version

1.10.0 (release)

### Companion app

GadgetBridge 0.70.0 (F-Droid)

Avamander, 2022-10-04:
If you want automatic reconnection, why aren't you enabling the specific option for it in Gadgetbridge?

InfiniTime can also *not* control if and when the companion decides to connect.

closed 2022-10-04 by Avamander

tinxx, 2022-10-04:
Wow, that was rough. Thanks for the feedback.