InfiniTimeOrg/InfiniTime #1355: Automatically reconnect to companion device

tinxx, 2022-10-04:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I want the watch to automatically reconnect to the companion app, e.g. GadgetBridge, after the connection is lost.

### Description

As described in #1354 other watches have the behavior of reconnection automatically.

Though it can possibly be forced by the companion app, I would prefer that the watch does it.

Avamander, 2022-10-05:
The watch with its current role can not initiate a connection in that direction. This is up to the companion.

closed 2022-10-05 by Avamander

tinxx, 2022-10-05:
Could you please be so kind as to elaborate why it is possible for Amazfit but not for InfiniTime to reinitiate the connection?

Avamander, 2022-10-05:
It can not. Device that fulfils the *central* role can connect to *peripherals*. *Peripheral* devices can advertise, they can not initiate a connection. That's just how BLE works.

tinxx, 2022-10-05:
I see, thank you. So they either implemented a workaround or the device-specific code in GadgedBridge lacks some finesse.

JF002, 2022-10-10:
As Avamander said, InfiniTime implements the peripheral role : it broadcast advertising messages (when it's not already connected to a central device). Then, host (central) device that receive those advertising messages can initiate a connection to the watch.

This seems to work pretty well on my side. I'm using Gadgetbridge on a quite recent Android version (12, I think).