InfiniTimeOrg/InfiniTime #1356: Error when building Infinitime

RazRosman, 2022-10-04:
I'm running the following make command:
make -j4 pinetime-app

and getting the following error:
[  0%] Built target infinitime_fonts_lv_font_navi_80
[  0%] Built target infinitime_fonts_lv_font_sys_48
[  1%] Built target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  1%] Built target infinitime_fonts_jetbrains_mono_42
[  1%] Built target infinitime_fonts_open_sans_light
[  1%] Built target QCBOR
[  1%] Built target infinitime_fonts_jetbrains_mono_76
[  1%] Built target infinitime_fonts_jetbrains_mono_bold_20
[  3%] Built target littlefs
[  8%] Built target infinitime_fonts
[ 32%] Built target lvgl
[ 42%] Built target nrf-sdk
[ 66%] Built target nimble
[ 66%] Building CXX object src/CMakeFiles/pinetime-app.dir/displayapp/screens/Screen.cpp.o
[ 66%] Building CXX object src/CMakeFiles/pinetime-app.dir/displayapp/DisplayApp.cpp.o
[ 66%] Building CXX object src/CMakeFiles/pinetime-app.dir/displayapp/screens/Clock.cpp.o
[ 66%] Building CXX object src/CMakeFiles/pinetime-app.dir/displayapp/screens/Tile.cpp.o
In file included from /home/pi/InfiniTime/src/displayapp/screens/Screen.cpp:1:0:
/home/pi/InfiniTime/src/displayapp/screens/Screen.h:12:7: error: template with C linkage
       template <class T> class DirtyValue {
       ^~~~~~~~
make[3]: *** [src/CMakeFiles/pinetime-app.dir/build.make:102: src/CMakeFiles/pinetime-app.dir/displayapp/screens/Screen.cpp.o] Error 1

RazRosman, 2022-10-05:
Using make -j pinetime-app instead of make -j4 pinetime-app solved the problem

but now I'm getting another error:

[ 91%] Building CXX object src/CMakeFiles/pinetime-app.dir/components/ble/CurrentTimeService.cpp.o
[ 91%] Building CXX object src/CMakeFiles/pinetime-app.dir/components/ble/DfuService.cpp.o
In file included from /usr/include/newlib/c++/7.3.1/utility:69:0,
                 from /usr/include/newlib/c++/7.3.1/array:38,
                 from /home/pi/InfiniTime/src/components/ble/BleController.h:3,
                 from /home/pi/InfiniTime/src/displayapp/screens/BleIcon.h:3,
                 from /home/pi/InfiniTime/src/displayapp/screens/BleIcon.cpp:1:
/usr/include/newlib/c++/7.3.1/bits/stl_relops.h:85:5: error: template with C linkage
     template <class _Tp>
     ^~~~~~~~

Avamander, 2022-10-05:
Use the correct compiler as specified in the build documentation.

closed 2022-10-05 by Avamander

Avamander, 2022-10-05:
If you look closely it tries to include headers from your system (for example "/usr/include/newlib/c++/7.3.1/bits/stl_relops.h"). That is in most cases, including this one, incorrect. 

This should not happen if you would use the ARM GCC compiler as specified in the build instructions.

RazRosman, 2022-10-06:
Which is?
I have gcc 9 installed.
I tried to install 10, but it didn't work.

JF002, 2022-10-10:
We currently build the project using ARM-GCC 10.3. See [this page](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildAndProgram.md) for more documentation about the build procedure. 

I recommend you download the exact same version of the compiler. You just need to download and unzip it on your system. Then, you just need to specify the path to the unzip folder with the CMake argument `ARM_NONE_EABI_TOOLCHAIN_PATH`.

Alternatively, you can use the [Docker container](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildWithDocker.md) to build the project.

JF002, 2022-10-10:
It looks like it cannot run `arm-none-eabi-gcc`. According to the build log, you're running on a raspberrypi. So you need to download the `aarch64` version of the toolchain. Note that this will only work if you run a 64bits OS on your raspberrypi. ARM does not seem to provide a ARM 32 bits version of the toolchain, unfortunately.

RazRosman, 2022-10-10:
Downloaded the aarch64 and still the same error.

What about using a different release of pinetime?

RazRosman, 2022-10-10:
This is the system I'm running

pi@raspberrypi:~/InfiniTime $ uname -a
Linux raspberrypi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
pi@raspberrypi:~/InfiniTime $ getconf LONG_BIT
32
pi@raspberrypi:~/InfiniTime $

JF002, 2022-10-11:
It seems that your rpi is running a 32 bits OS, so it won't be able to run the aarch64 version of the toolchain. I guess you have 2 options : install a 64 bits OS on your rpi or find (or build yourself) a 32 bits version of the gcc-arm-none-eabi toolchain.

Avamander, 2022-10-11:
> are you familiar with a 32 bit version?

Does not officially exist, 32bit ARM Linux very is just quite legacy.

> Isn't that the version mentioned in requirements of this article is 32 bit? what is wrong with that?

Not sure what you mean by this,

> Can you direct me to an article for building the toolchain?

The source tarball will contain those instructions, but it's not guaranteed to work especially if there's no official build either.

> Is this going to be the same issue with all pinetime releases?

No, because it's an issue with an OS edition being obsolete. Yes, because you'll need something more modern to reduce hassle.

JF002, 2022-10-11:
I only work on AMD64/X86-64 and AARCH64 (ARM 64 bits), so no, I unfortunately won't be able to help you install or build a toolchain for your OS. 
The version of the PineTime and/or InfiniTime won't change anything : your OS is unable to run the compiler needed to build the firmware.