InfiniTimeOrg/InfiniTime #1357: Building pinetime-app raised an error

RazRosman, 2022-10-05:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Error when running "make -j pinetime-app"

### What should happen instead?

Building Successfully

### Reproduction steps

I follow the instructions for building the project.
I ran: cmake -S . -B build -DARM_NONE_EABI_TOOLCHAIN_PATH=/usr -DNRF5_SDK_PATH=/home/pi/Downloads/SDK/nRF5_SDK_17.1.0_ddde560

that finished successfully.

But when running make -j pinetime-app

I get the following error:

[ 91%] Building CXX object src/CMakeFiles/pinetime-app.dir/components/ble/CurrentTimeService.cpp.o
[ 91%] Building CXX object src/CMakeFiles/pinetime-app.dir/components/ble/DfuService.cpp.o
In file included from /usr/include/newlib/c++/7.3.1/utility:69:0,
                 from /usr/include/newlib/c++/7.3.1/array:38,
                 from /home/pi/InfiniTime/src/components/ble/BleController.h:3,
                 from /home/pi/InfiniTime/src/displayapp/screens/BleIcon.h:3,
                 from /home/pi/InfiniTime/src/displayapp/screens/BleIcon.cpp:1:
/usr/include/newlib/c++/7.3.1/bits/stl_relops.h:85:5: error: template with C linkage
     template <class _Tp>
     ^~~~~~~~

That crashes the make process.

### More details?

Searched the web for a solution.
Tried to install different NRF5 SDK versions.

Another info that may help.

I'm using the following gcc

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/gcc-9.1.0/libexec/gcc/arm-linux-gnueabihf/9.1.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../gcc-9.1.0/configure --prefix=/opt/gcc-9.1.0 --build=x86_64-pc-linux-gnu --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-multilib --program-suffix=-9.1
Thread model: posix
gcc version 9.1.0 (GCC) 

### Version

using the one cloned from here: "git clone https://github.com/InfiniTimeOrg/InfiniTime.git"

### Companion app

I'm not. Using rpi3 connected with a wire through the rpi pins to PIneTime

FintasticMan, 2022-10-05:
As Avamander already stated in the last issue you opened, you need to use the GCC toolchain specified in the build instructions.

Avamander, 2022-10-05:
Duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/1356

closed 2022-10-05 by Avamander