InfiniTimeOrg/InfiniTime #1362: Vibration length/strength adjustment

thelunarfox75, 2022-10-08:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Something in the ui to change the vibration length / strength

### Description

last time something like this was posted was a year ago, a response was given telling OP they would be working on something like that and i just wanted some kind of update

I use pinetime with infinitime to help remember to take blood glucose tests, but i've kinda gotten used to the vibrations and wanted to make them a bit longer

SteveAmor, 2022-10-09:
Duplication of #210

minacode, 2022-10-09:
There is some adjustment (and place for discussion) in #1328.

Riksu9000, 2023-02-20:
Closing as a duplicate of #210 or #391

closed 2023-02-20 by Riksu9000