InfiniTimeOrg/InfiniTime #1363: Flashlight changes brightness in specific circumstance

crmcalls, 2022-10-10:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Flashlight app changes brightness to high after closing

### What should happen instead?

Brightness should return to the brightness set by user instead of high

### Reproduction steps

- Set brightness to low or medium
- Open flashlight app
- Return to watchface by **holding** down side button

Brightness will be changed from low or medium to high

### More details?

Noticed and wanted to report it, as I can't fix it myself.

### Version

1.10.0

Joheyy, 2022-10-10:
Interesting that this doesn't affect clicking the button. Any other event I tested produces this bug. (Long, very long and double press, then timer finish and also notifications) 

This shouldn't be the case since every one of these events call LoadApp which itself should reset the brightness here:
https://github.com/InfiniTimeOrg/InfiniTime/blob/964f065de43bf2e5b681d9ca3ebc1e335843a717/src/displayapp/DisplayApp.cpp#L306

Only difference I can see is that ButtonPushed calls LoadPreviousScreen instead of directly calling LoadApp. I can't see why this matters tho since LoadPreviousScreen only calls LoadApp itself. So maybe a timing issue? 

More research is required.

minacode, 2022-10-11:
`ApplyBrightness` has this weird check for invalid data where it defaults to High. I don't think that this is the bug, but we should keep it in mind. 
Also I think that this behavior should be done inside the settingsController if necessary.

Joheyy, 2022-10-11:
After actually testing the newest version I noticed that this is already fixed. 

This issue is a duplicate of #1201 and was fixed by #1213

closed 2022-10-16 by Riksu9000