InfiniTimeOrg/InfiniTime #1364: Streaming the data (real-time) from PineTime watch to PC/laptop with InfiniTime?

stevensu1838, 2022-10-11:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Streaming the data (real-time) from PineTime watch to PC/laptop  with InfiniTime?

### Description

Hi team

I am a researcher from an university, and we've been working on a project aiming to stream live data (such as heart rate data) from Pine Time watch to the PC/laptop. I've learned that the InfiniTime might be a good way to make our goal happen. May I please ask if streaming the data (real-time) from PineTime watch to PC/laptop is doable with InfiniTime? If so, I will give it a try. Cheers

minacode, 2022-10-11:
The companion [itd](https://gitea.arsenm.dev/Arsen6331/itd)
can do that for you. Specifically for heart rate I would not trust the sensor for scientific purposes though, but that is only my take. It depends on what you want to do with it. 
The motion values are coming directly from the hardware (Bma421 or Bma425) and should have a good quality.

JF002, 2022-10-11:
Here are a few additional info to @minacode's answer.

InfiniTime exposes a few data via BLE:
 - Battery level
 - Heart rate value (when enabled)
 - Step count
 - Motion (x/y/z accelerations)

The companion app can read those data (polling) or subscribe to the corresponding BLE characteristics to receive a notification each time the value changes (using the BLE NOTIFY functionality).

stevensu1838, 2022-10-17:
Hi @minacode and @JF002,

Thank you so much for your great help. I am very new to PineTime and InfiniTime. I know how to program though. My goal is simply to stream the data (real-time) from the PineTime watch to PC. And I've looked into both the InfiniTime document and the companion [itd](https://gitea.arsenm.dev/Arsen6331/itd). May I please ask if can I directly start my project using the companion [itd](https://gitea.arsenm.dev/Arsen6331/itd) alone or do I need to do something with InfiniTime first? Are the two independent of each other? My last question is do you have any document/tutorial more beginner friendly to start with a PineTime Watch development kit? I find InfiniTime fascinating but have lost my direction in browsing the official document. I haven't been able to do anything with InfiniTime and my PineTime Watch development kit. Which should be the very first step? Thank you both for everything. Best regards

minacode, 2022-10-17:
> May I please ask if can I directly start my project using the companion itd alone or do I need to do something with InfiniTime first?

It's very simple. 
1. Install itd on your linux. There are packages for the main distributions.
2. Start it. This triggers the watch to pair with your PC.
3. Run itctl to get the data you want.
4. In case of heart rate data you have to start the measurement on the watch by pressing the button in the heart rate app.

> Are the two independent of each other?

Itd is not a project of InfiniTimeOrg, but itd exists only for InfiniTime and the author of itd is part of the itd community. Itd is not the only companion on Linux but it is the best supported one. There seems to be active communication between them. @JF002 can probably say this better.

JF002, 2022-10-17:
> May I please ask if can I directly start my project using the companion [itd](https://gitea.arsenm.dev/Arsen6331/itd) alone or do I need to do something with InfiniTime first?

It mostly depends on what you want to do. If InfiniTime already exposes the data you want to stream (steps, HR, battery), you don't need to change anything to InfiniTime. Same goes for the companion apps : ITD (but also Amazfish, Siglo, Gadgetbridge,...) already support most of the functionality from InfiniTime, so you probably won't need to do anything for them to work with InfiniTime. 
On the other side, if you want to stream other data, or change the acquisition rate or anything else, so yes, you'll need to work in InfiniTime and/or the companion app.

> Are the two independent of each other?

InfiniTime and the companion apps are effectively independent projects, but we all work together to create a nice ecosystem around the PineTime. There are multiple companion apps that currently support the PineTime. Some of them also support other devices:
 - [ITD](https://gitea.arsenm.dev/Arsen6331/itd)
 - [Amazfish](https://github.com/piggz/harbour-amazfish)
 - [Gadgetbridge](https://codeberg.org/Freeyourgadget/Gadgetbridge/)
 - [InfiniLink](https://github.com/InfiniTimeOrg/InfiniLink)
 - [Watchmate](https://gitlab.com/azymohliad/watchmate)
 - [Siglo](https://github.com/theironrobin/siglo)

> My last question is do you have any document/tutorial more beginner friendly to start with a PineTime Watch development kit? I

There are quite a lot of documentation available but it's probably a bit scattered around the web. Here are a few links
 - [The Wiki](https://wiki.pine64.org/wiki/PineTime) : contains a lot of info about the PineTime. You'll find the datasheets, schematics, documentation on wiring the devkits and many links.
 - [Lup Yuen Lee](https://lupyuen.github.io/) : Lup wrote **a lot** of articles about the PineTime. They are a goldmine of information!
 - [The documentation of InfiniTime](https://github.com/InfiniTimeOrg/InfiniTime#new-to-infinitime) : we do our best to maintain the documentation up to date. It contains info about the code, how to build and flash the PineTime,...
 - You can also have a look at a [quick started guide](https://infinitime.io/blog/2020-09/i-received-my-pinetime-devkit-now-what/) I wrote when we released InfiniTime 0.7.1 (the first version that was flashed by default on the PineTime)

>  I find InfiniTime fascinating...

Thanks!

> ...but have lost my direction in browsing the official document

Yeah, I understand. We are currently trying to [re-work the documentation](https://github.com/InfiniTimeOrg/InfiniTime/pull/838) but this is progressing slowly...

>  I haven't been able to do anything with InfiniTime and my PineTime Watch development kit. Which should be the very first step?

The first step would probably be to setup a [development environment](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildAndProgram.md) (or [with docker](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildWithDocker.md) for InfiniTime and/or for the companion app of your choice (refer to the project documentation). 

You can also use [InfiniSim](https://github.com/InfiniTimeOrg/InfiniSim), the InfiniTime simlator to do some development. It mostly simulate the UI, and not the BLE connectivity, though.

Then, if you have the devkit, you will also want to set it up with your SWD probe ([here](https://wiki.pine64.org/wiki/PineTime_Devkit_Wiring) and [here](https://infinitime.io/blog/2021-03/my-new-pinetime-development-setup-with-current-measurement/)).
And finally flash your devkit [here](https://wiki.pine64.org/wiki/Reprogramming_the_PineTime), [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/openOCD.md), [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/SWD.md), [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/jlink.md) and [here](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/gdb.md).

stevensu1838, 2022-10-18:
@minacode ,
Hi, thank you so much for your great help. And I've been trying out the itd by following you instruction above and the [itd document](https://gitea.arsenm.dev/Arsen6331/itd).
I've successfully install itd-bin on my Arch Linux PC by using yay -S itd-bin. In the terminal, it shows it is installed as below:  

`[lui@crossbow ~]$ yay -Ss itd-bin
aur/itd-bin 0.0.8-1 (+1 1.00) (Installed)
    Companion daemon for the InfiniTime firmware on the PineTime smartwatch`

My question is how do I start itd to trigger the watch to pair with my PC? I am sorry I am new to linux. I did something below but if didn't work:
`[lui@crossbow ~]$ {"Receiver": "ITD", "Method": "Notify", "Arg": {"title": "title1", "body": "body1"}, "ID": "some-id-here"}
bash: {Receiver:: command not found
[lui@crossbow ~]$ itctl
NAME:
   itctl - A new cli application

USAGE:
   itctl [global options] command [command options] [arguments...]

COMMANDS:
   help            Display help screen for a command
   resources, res  Handle InfiniTime resource loading
   filesystem, fs  Perform filesystem operations on the PineTime
   firmware, fw    Manage InfiniTime firmware
   get             Get information from InfiniTime
   notify          Send notification to InfiniTime
   set             Set information on InfiniTime
   update, upd     Update information on InfiniTime
   watch           Watch a value for changes

GLOBAL OPTIONS:
   --socket-path value, -s value  Path to itd socket (default: "/tmp/itd/socket")
[lui@crossbow ~]$ 
`
Can you please help? How do I start runing itd and how do I use the cmmand itctl to get the data you want. Thank you so much

minacode, 2022-10-18:
```
systemctl start --user itd.service
```
starts it. 

```
systemctl enable --user itd.service
```
makes it start every time when booting. So you have to do a separate start the first time. 

See [here](https://gitea.arsenm.dev/Arsen6331/itd#starting). 

A prompt should appear on your linux and a code on your watch. 
Enter that code. 
If you ever have connection issues, remove the watch from your paired devices on your linux. That will trigger this prompt again. 
The watch should show a notification upon successful (re)connection. 
You can then use `itctl`.

stevensu1838, 2022-10-19:
@minacode 
Hi, Thank you so much for your great help. With your instruction, I successfully connected the PC to the watch as shown in the pic below:  

[image]

However, no prompt appears on my Linux and no code on your watch. Any idea? Thank you so much again.

minacode, 2022-10-19:
Well, seems like you are connected 😉 Does itctl work? 
I guess you have an older firmware that does not support pairing (which is no problem in itself). You can [update](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/gettingStarted/updating-software.md) it via itctl. Follow the whole procedure, including validating the firmware and updating the resources.

stevensu1838, 2022-10-20:
@minacode,
Hi thank you so much for your help. You are amazing. itcl works for me as shown in the code blow: 
`[lui@crossbow ~]$ systemctl --user stop itd
[lui@crossbow ~]$ systemctl --user start itd
[lui@crossbow ~]$ itctl get
NAME:
   itctl get - Get information from InfiniTime

USAGE:
   itctl get command [command options] [arguments...]

COMMANDS:
   address, addr  Get InfiniTime's bluetooth address
   battery, batt  Get InfiniTime's battery percentage
   heart          Get heart rate from InfiniTime
   motion         Get motion values from InfiniTime
   steps          Get step count from InfiniTime
   help, h        Shows a list of commands or help for one command

OPTIONS:
   --help, -h  show help (default: false)

[lui@crossbow ~]$ itctl get heart
0 BPM
[lui@crossbow ~]$ itctl get heart
122 BPM
[lui@crossbow ~]$ itctl get heart
155 BPM
[lui@crossbow ~]$ 

`
I can successfully read the heart rate data from the watch, and my firmware version is 1.6.0. So the only problem is no prompt appears on my Linux and no code on my watch. Do I still need to update the firmware to the latest version? 

My last question is I'd love to display the heat rate data in number and line graph trough the [itgui](https://gitea.arsenm.dev/Arsen6331/itd#itgui) as shown in the picture below, however the itgui document is hard to follow, could you please elaborate a bit about how to use the itgui command? Thank you from the bottom of my heart. Cheers 

[image]
[image]

stevensu1838, 2022-10-20:
@minacode 

To be clear, I find the following instruction from itgui hard to follow, could you please make it beginner friendly. Like where is cmd/itgui. Really appreciate it 

itgui

In cmd/itgui, there is a gui frontend to the socket of itd. It uses the [Fyne library](https://fyne.io/) for Go.
Compilation

Before compiling, certain prerequisites must be installed. These are listed on the following page: https://developer.fyne.io/started/#prerequisites

It can be compiled by running:

go build ./cmd/itgui

Cross-compilation

Due to the use of OpenGL, cross-compilation of itgui isn't as simple as that of itd and itctl. The following guide from the Fyne website should work for itgui: https://developer.fyne.io/started/cross-compiling.

FintasticMan, 2022-10-20:
You should be able to just run itgui on the command-line and get that gui running. If you want other things such as the motion data sent to your computer, you will need to update your watch to the latest firmware. This can be done using itctl by downloading the 2 zip files from [the GitHub release](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.11.0), and running `itctl fw upg -a pinetime-mcuboot-app-dfu-1.11.0.zip -r infinitime-resources-1.11.0.zip`.

minacode, 2022-10-20:
`itgui` is not part of the Arch package. You can download it by running in the path of your choice:
```
git clone https://gitea.arsenm.dev/Arsen6331/itd.git
```
Then move into the dir
```
cd itd
```
and build according to the docs you cited with
```
go build ./cmd/itgui
```
Then run it with 
```
cmd/itgui
```
which is basically just the path to it, depending on where you currently are.

If this fails it might be due to me not testing this while typing or because you need to install the `go` package on your Arch first.

Also I would always recommend to install updates. Why stay on an older version?

stevensu1838, 2022-10-21:
@minacode 
Hi I followed your steps and there is no errors. However, I got the problem at the last step when I key in cmd/itgui. It shows bash: cmd/itgui: Is a directory

[image]

I really appreciate your great support. You are so warm-hearted. And we are getting there

stevensu1838, 2022-10-21:
@FintasticMan Hi, thank you so much for your input. Just let you know that  just run itgui on the command-line doesn't get that gui running somehow
[image]

minacode, 2022-10-21:
> If this fails it might be due to me not testing this while typing

:grinning: 

After building itgui with go the binary is located in the itd directory. You can run
```
./itgui
```
from there or
```
itd/itgui
```
from your home as it seems.

`itgui` is not something that your command line will magically know. This has to do with the PATH variable and is a whole other topic. (Tldr: your shell knows everything it can find in one of the directories that your path variable contains. Your itd directory is not one of those places but you could add it if you want).

stevensu1838, 2022-10-22:
Hi@minacode, you are the legend and thanks for the explanation. I've brought out the gui but it throws errors as shown below. Do you know why? Thanks a lot in advance. Cheers
[image]

FintasticMan, 2022-10-22:
I think that might be an issue because your watch is on an older version of InfiniTime, you can update it to the latest version using `itctl fw upg -a ~/Downloads/pinetime-mcuboot-app-dfu-1.11.0.zip`, and when it has finished rebooting, running `itctl res load ~/Downloads/infinitime-resources-1.11.0.zip`. You can get the zip files needed for those commands from [the GitHub releases page](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.11.0).

stevensu1838, 2022-10-23:
Hi @FintasticMan, 

I've been trying to update the version of InfiniTime, but it throws the following errors. And I didn't find detailed instructions on how to use itctl to update the version of firmware. Could you please tell me more on how to do it? Thanks a lot

[image]

stevensu1838, 2022-10-23:
Hi @FintasticMan, I successfully updated the version of InfiniTime using itctl fw upg -a ~/Downloads/pinetime-mcuboot-app-dfu-1.11.0.zipusing, and then I rebooted the watch and validated the new firmware on the watch. However it throws the following errors when I running itctl res load ~/Downloads/infinitime-resources-1.11.0.zip
running itctl res load ~/Downloads/infinitime-resources-1.11.0.zip

stevensu1838, 2022-10-23:
> 

Hi @minacode, after I updated the version of InfiniTime to 1.11.0. Now  a code does show on my watch, however, no prompt appear on my linux. Is there anything that I can do?

minacode, 2022-10-23:
For your screenshots: double check the file names or use the tab key to autocomplete while typing.

The missing socket error appears when there is no connection.

On the watch you can confirm your firmware version in the settings under "Firmware". 

The prompt not showing seems to be a problem of your Linux. I have no good solution for this. Maybe wait some seconds? It's not always instant but should be fast enough.

FintasticMan, 2022-10-23:
Maybe you could try unpairing your watch in your computer's bluetooth settings, and then pairing again. That is necessary sometimes when updating from a version without secure pairing.

minacode, 2022-10-23:
Regarding your Ubuntu:

> See user logs and 'systemctl --user status itd.service' for details.

When your shell gives you such hints, consider using them. What is the status of this service?

> It is much hard than I expected to use and hopefully our conversations here can help other beginers. 

Don't take this personally but I feel like you have the worst luck in this process 😉 
But yes, hopefully we can get a good documentation out of this in the end.

stevensu1838, 2022-10-23:
@minacode, please check out the user logs and 'systemctl --user status itd.service' for details for the Ubuntu machine
[image]

minacode, 2022-10-23:
Yes, there is obviously something wrong. It is red and called 'error' :grinning: 
Best is that you ask that question over at the itd repository I linked to previously. You will find people that can solve this problem there far more likely. I can not debug the itd systemd unit on your Ubuntu for you because I do not use Ubuntu.

Have you tried unpairing your watch from the Arch as @FintasticMan said? This seems to be a good idea.
[Here](https://askubuntu.com/questions/758586/how-to-unpair-bluetooth-device-from-the-command-line) is what I found from a quick search on how to do that.

stevensu1838, 2022-10-24:
Hi @minacode and @FintasticMan, I've tried unpairing my watch from Arch Linux and the situation is still the same: A code does show on my watch, however, no prompt appear on my linux. Thank you both so much.

stevensu1838, 2022-10-24:
@minacode @FintasticMan @JF002 , Hi all, as you know I simply want to stream the heart rate data from the PineTime watch to my PC, do you have other ways to do it instead of itd? Sorry, haven't got that running yet. Also, I've noticed that the heart rate data is not that accurate from the watch. Is it caused by the sensor itself or the coding/algorithm? If it is caused by the hardware, I might have to move to other smart watches. Thank you all so much. Cheers

minacode, 2022-10-24:
> the situation is still the same: A code does show on my watch, however, no prompt appear on my linux.

Can you run 
```
itd
```
manually on a command line? You may have to 
```
systemctl stop --user itd.service
```
before that. You should be able to enter the code on the command line then. If not we might get some output regarding errors.
In a second terminal you can then use `itctl`.

> Also, I've noticed that the heart rate data is not that accurate from the watch. Is it caused by the sensor itself or the coding/algorithm?

> Specifically for heart rate I would not trust the sensor for scientific purposes though, but that is only my take. It depends on what you want to do with it.

The PineTime costs around $27. You won't get AppleWatch-level measurements.

stevensu1838, 2022-10-25:
@minacode Thank you so much for everything. Is there any chance you can remote control my linux PC and take a look at what's going on? Can we use teamviewer on Arch Linux? And please check out the logging below too see if you can find some information useful. Cheers

`[lui@crossbow ~]$ systemctl --user stop itd
[lui@crossbow ~]$ ITD_LOGGING_LEVEL=debug itd
10:40PM DBG No device found in list, attempting to discover
10:40PM DBG InfiniTime device discovered mac=F7:58:76:0E:DF:CE
10:40PM DBG Device not connected, connecting
WARN[0000] Recovering from panic: send on closed channel 
10:40PM DBG Device not paired, pairing
10:40PM DBG Passkey requested, calling onReqPasskey callback
Passkey: 09510:40PM FTL Error connecting to InfiniTime error="Authentication Canceled"
[lui@crossbow ~]$ systemctl --user stop itd
[lui@crossbow ~]$ ITD_LOGGING_LEVEL=debug itd
10:42PM DBG No device found in list, attempting to discover
10:42PM DBG InfiniTime device discovered mac=F7:58:76:0E:DF:CE
10:42PM DBG Device not connected, connecting
10:42PM DBG Device not paired, pairing
10:42PM DBG Passkey requested, calling onReqPasskey callback
Passkey: 730006
10:42PM DBG Resolved characteristic name="Step Count" uuid=00030001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name="Motion Values" uuid=00030002-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name="New Alert" uuid=00002a46-0000-1000-8000-00805f9b34fb
10:42PM DBG Resolved characteristic name= uuid=00000002-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name= uuid=00000003-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name="Notification Event" uuid=00020001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name="Current Time" uuid=00002a2b-0000-1000-8000-00805f9b34fb
10:42PM DBG Resolved characteristic name="Battery Level" uuid=00002a19-0000-1000-8000-00805f9b34fb
10:42PM DBG Resolved characteristic name= uuid=00001531-1212-efde-1523-785feabcd123
10:42PM DBG Resolved characteristic name= uuid=00000001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name= uuid=00001532-1212-efde-1523-785feabcd123
10:42PM DBG Resolved characteristic name="Heart Rate" uuid=00002a37-0000-1000-8000-00805f9b34fb
10:42PM DBG Resolved characteristic name="Filesystem Version" uuid=adaf0100-4669-6c65-5472-616e73666572
10:42PM DBG Resolved characteristic name= uuid=00000004-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Resolved characteristic name="Filesystem Transfer" uuid=adaf0200-4669-6c65-5472-616e73666572
10:42PM DBG Resolved characteristic name="Firmware Version" uuid=00002a26-0000-1000-8000-00805f9b34fb
10:42PM DBG Resolved characteristic name= uuid=00000005-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Firmware Version" uuid=00002a26-0000-1000-8000-00805f9b34fb
10:42PM INF Connected to InfiniTime version=1.11.0
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="New Alert" uuid=00002a46-0000-1000-8000-00805f9b34fb
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Current Time" uuid=00002a2b-0000-1000-8000-00805f9b34fb
10:42PM INF Initialized InfiniTime music controls
10:42PM INF Relaying notifications to InfiniTime
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Filesystem Transfer" uuid=adaf0200-4669-6c65-5472-616e73666572
10:42PM INF Started control socket path=/tmp/itd/socket
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Adding weather event event={"DewPoint":2,"EventType":3,"Expires":3600,"Temperature":1110,"Timestamp":1666604552}
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Adding weather event event={"Amount":0,"EventType":1,"Expires":3600,"Timestamp":1666604552,"Type":0}
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Adding weather event event={"DirectionMax":10,"DirectionMin":10,"EventType":2,"Expires":3600,"SpeedMax":1,"SpeedMin":1,"Timestamp":1666604552}
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Adding weather event event={"Amount":0,"EventType":8,"Expires":3600,"Timestamp":1666604552}
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Adding weather event event={"EventType":9,"Expires":3600,"Humidity":54,"Timestamp":1666604552}
10:42PM DBG Checking characteristic status
10:42PM DBG Characteristic available name="Weather Data" uuid=00040001-78fc-48fe-8e23-433b3a1942d0
10:42PM DBG Adding weather event event={"EventType":6,"Expires":3600,"Pressure":1017,"Timestamp":1666604552}

`

minacode, 2022-10-25:
Looks like it worked. You even inserted pairing keys. And then it connected to various services. I cannot read that something went wrong.

Did you try?
> In a second terminal you can then use `itctl`.

stevensu1838, 2022-10-25:
@minacode I get it working with your instructions.
Terminal 1
[image]
Ternimal 2
[image]
Gui
[image]

The only problem is the watch keeps showing new numbers for pairing and It throws a bit error as you can see in the picture above. I just ignore these and it's been working. 

Another question is I hope to see the line graph as show below. However, there is no **Metrics** option for me. Could you please help? 

[image]

minacode, 2022-10-25:
> WRN Multiple connection attempts have failed. If this continues, try removing the InfiniTime device from bluetooth

This can happen sadly. Removing the pairing and starting itd again should be enough for a stable connection. 

> Another question is I hope to see the line graph as show below. However, there is no Metrics option for me. Could you please help?

I can't answer that right now, sorry.

But you can stream the data in the terminal, right?

Bonus knowledge: 
```
cmd > file
```
writes the output of `cmd` into a file called `file`. Change both such that it fits your needs. You can use that to easily capture the streamed data into a file.

stevensu1838, 2022-10-25:
@minacode 

Thank you so for for everything. especially for this bonus knowledge. I just tried it but didn't get it working. Did I do something wrong? What does cmd do? I do need capture the streamed data into a file. Thanks a million.

[image]

stevensu1838, 2022-11-10:
@minacode 
I am still trying to save data and plot it myself. But when I type in **itctl get heart >** in the terminal, it throws syntax errors.  I only need to read and plot the time steps and heart rate data to make a very simple plot. Could you please elaborate on how I can save the heart rate data into a csv, excel or any other file? I am counting on you. Thanks a lot

May I please ask how shuld I open and read the db file correcty on arch linux? Thanks a lot. Cheers

FintasticMan, 2022-11-10:
Have a look into Linux shell redirection and piping, that should give you an idea on how to write the output from a command to a file. You might also want to use something like `sed` or `awk` to modify the output from the command.

minacode, 2022-11-10:
There is `itctl watch` that lets you monitor a value repeatedly? It outputs the current values until you stop it. Each value is written on a new line. Putting that into a file like I described above (`>`) gives you a file with one value per line. That is already a valid [csv file](https://en.wikipedia.org/wiki/Comma-separated_values). 
If you name it with an `.csv` ending you should be good.
Timestamps are difficult, because the watch does not make them. You can trust the watch to measure at constant intervals at least as much as you can trust the heart for scientific purposes. Maybe that is enough for you to male your own time stamps.

stevensu1838, 2022-11-11:
@minacode you are such an inspiring person. it's so cool to know so much about everything. Do you know what exactly the heart rate data logging interval is from the PineTime watch? Can you please show the heart rate interval part of the current algorithm? I'd love to modify the logging frequency. Cheers

minacode, 2022-11-11:
I was wrong. InfiniTime sends a new heart rate once the value changes. That is definitely not in regular intervals. 
A quick search found me [this link](https://stackoverflow.com/questions/21564/is-there-a-unix-utility-to-prepend-timestamps-to-stdin) where the best solution is to install `moreutils` on your Linux to gain access to the program `ts` that can timestamp data. The timestamps would then be added when the measurements arrive on your computer.

What you probably want is:
```
itctl watch heart | ts "%F, %T," > data.csv
```
where `data.csv` can be any file name you want.

You can change the output format which above is `"%F, %T,"` to whatever you want with [this link](https://linux.die.net/man/3/strftime). Note the commas. They divide your cells in the csv format.

I learned something from this :smile:

stevensu1838, 2022-11-14:
Hi @minacode, thank you so much for your effort. As you said, InfiniTime sends a new heart rate once the value changes. That is definitely not in regular intervals. If updating is not in regular interval, how can I make a plotting of the heart rate data as the one shown below? Any idea, please?

[image]

minacode, 2022-11-14:
You are getting *every* value change with that method. That should be enough.

Anachron, 2022-11-14:
@stevensu1838 you could pipe the itctl values into a file and read it regulary (30s intervals, for example) and pipe this again into another file.

Anachron, 2022-11-15:
True, but depending on how the plot is made he may need the additional data points with unchanged values.

stevensu1838, 2022-11-15:
@Anachron Thank you both so much. I want the time interval every 1s or every 5s or every 30s. I do think I need the additional data points with unchanged values to make the plotting continuous. I am trying out your will. And can you please say a bit more about how  1) read the file regulary (30s intervals, for example) and 2) pipe this again into another file. I've never done this before. Cheers

stevensu1838, 2022-11-15:
@minacode @Anachron You guys are amazing! I've record the data with timestamps as @minacode instructed. And I am trying to read it regularly (5s intervals, for example) and pipe this again into another file. Please let me know how I can add the additional data points with unchanged values to fill some of the empty value while reading the file regularly. You guys are awesome. I learn a lot from you all

[image]

minacode, 2022-11-15:
Can you just rename your file to a `.csv` ending and open it in the table program of your choice? I don't get why you need the additional values.

And don't name your files`.txt`. That makes interpretation of them from programs harder. They will handle the file according to the ending. Your comment is not *text* (`.txt`) but a table which is `.csv`.

stevensu1838, 2022-11-15:
@Anachron @minacode ,

Please find below an example of the two sets of data that I want to plot in one figure and compare. Please let me know if you have any good ideas. Cheers.

[image]

minacode, 2022-11-15:
> I am a researcher from an university

Are you sure? Sorry, but come on... Every technical help around InfiniTime aside, but if that is what you are researching then you should be able to figure out how to plot two graphs in the same diagram. I am not going to do your job.

stevensu1838, 2022-11-15:
@minacode Hi I think you misunderstood me, I will keep exploring and come back to you. Don't get pissed off by stupid questions. I've learnt a lot from you, you've helped me a lot and I hope you will be happy. Anyway, Let me think about it and come back to you soon. Be happy

minacode, 2022-11-15:
Don't get me wrong. I gladly helped you and don't believe in "stupid questions". It's no problem to be inexperienced with Linux and such. We all start somewhere.

> How do I plot two data sets, one with regular intervals and the other one with the irregular interval, in the same plotting to compare? 

But I don't want to plot or interpret your numbers for you. That is not a technical problem. And its not even difficult. You know that the PineTimes values are constant between the data points, so you can perfectly estimate them.

stevensu1838, 2022-11-15:
@minacode Hi Thanks a lot for your understanding. Please forgive me if I ask the following question wrong. I am plotting the PineTime data your helped me saved and I am plotting it in matlab. I find this [example on plotting very similar data](https://www.mathworks.com/matlabcentral/answers/286722-how-can-i-plot-time-series-data-from-csv). 
Their data 
[image]
Our PineTime data
[image]

I can use the their code to plot their example data with no problem.
[image]

However, when I switch to our data PineHeartOutput.csv, it throws the following errors:

[image]

I don't really understand our data is the same as the example data. The only difference is we don't have AM or PM in the end of timestamps. Sorry if you don't even use matlab. Just feel like you know everything. Thanks a lot for all your help

minacode, 2022-11-15:
I am not familiar with Matlab, sorry. The error says that the `2` in `y = d(:,2)` is too big. I suspect the second column in d is missing. Maybe `xlsread` is wrong because you do not have an`.xls` but a `.csv`. You could try `csvread` instead, but I am just guessing.

BenjaminMickler, 2022-11-25:
In case you are familiar with Python, here's a snippet from a project that I am working on that reads heart rate data from InfiniTime using a library called bleak (https://github.com/hbldh/bleak). It won't work if the watch is already connected to a device. This script will handle the connection. Of course you would want to remove all of the seizure monitor related stuff. Don't worry about the copyright or the licence, you're welcome to use the code however you like.

```Python
__author__ = "Benjamin Mickler"
__copyright__ = "Copyright 2022 Benjamin Mickler"
__credits__ = ["Benjamin Mickler"]
__license__ = "GPLv3 or later"
__version__ = "20221126"
__maintainer__ = "Benjamin Mickler"
__email__ = "ben@benmickler.com"

"""
This file is part of The Seizure AI Project.

The Seizure AI Project is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by the Free
Software Foundation, either version 3 of the License, or (at your option) any
later version.

The Seizure AI Project is distributed in the hope that it will be useful, but
WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more
details.

You should have received a copy of the GNU General Public License along with
The Seizure AI Project. If not, see <https://www.gnu.org/licenses/>.
"""

import asyncio
from bleak import BleakClient, BleakScanner
import requests
import json
import signal
import traceback

# MAC address: ff:1c:2a:bf:8a:3b

print("A small neural network to try to detect a seizure in 10 heart rate samples. All predictions are done on the server. Please use sparingly as the server only has 1GB of RAM and may not be able to make many predictions at once. It's only an experiment, don't rely on it to alert you to seizures. Copyright © Benjamin Mickler 2022\n")

threshold = input("Threshold (blank for 0.8): ")
if threshold == "":
    threshold = 0.8
else:
    threshold = float(threshold)

name = input("Heart rate monitor name (blank to use InfiniTime): ")
if name == "":
    name = "InfiniTime"
def get_mac_address(name):
    print(f"Searching for {name}")
    from bleak import discover
    devices = asyncio.run(BleakScanner.discover())
    for d in devices:
        if d.name == name:
            return d.address
    return None

readings = []

_models_response = requests.get("https://seizure.ai.benmickler.com/models")
_models = json.loads(_models_response.json())
models = {}

for i, model in enumerate(_models, start=1):
    models[i] = model

service_uuid = "0000180d-0000-1000-8000-00805f9b34fb"
char_uuid = "00002a37-0000-1000-8000-00805f9b34fb"

async def run(address, loop):
    disconnected_event = asyncio.Event()
    def disconnected_callback(client):
        print("Device disconnected")
        disconnected_event.set()
    client = BleakClient(address, loop=loop, disconnected_callback=disconnected_callback)
    try:
        await client.connect()
        print(f"Connected: {client.is_connected}")
        await client.start_notify(char_uuid, callback)
        await disconnected_event.wait()
    except Exception as e:
        #print(traceback.format_exc())
        pass
    finally:
        if client.is_connected:
            print("\nDisconnecting")
            await client.stop_notify(char_uuid)
        await client.disconnect()

def callback(sender, data):
    print("Received: {0}".format(data[-1]))
    readings.append(data[-1])
    if len(readings) >= 10:
        print()
        r = requests.post("https://seizure.ai.benmickler.com/predict", json={"model_name": model, "heart_rate": readings})
        request_json = json.loads(r.json())
        if "error" in request_json:
            print(f"Error: {request_json['error']}")
        else:
            print(f"Result: {request_json['result']}")
            print(f"Seizure: {'yes' if float(request_json['result']) >= threshold else 'no'}")
        readings.pop(0)

if __name__ == "__main__":
    print("Models:")
    for k, v in models.items():
        print(f"{k}: {v}")
    model_input = int(input("Model: "))
    if model_input not in models:
        print("Invalid model")
        exit(1)
    model = models[model_input]
    loop = asyncio.new_event_loop()
    address = get_mac_address(name)
    if address is None:
        print(f"Could not find device with name {name}")
        address = input("Please enter MAC address: ")
    asyncio.run(run(address, loop))```

JF002, 2022-11-30:
Let's continue this post in [Discussions](https://github.com/InfiniTimeOrg/InfiniTime/discussions) :-)