InfiniTimeOrg/InfiniTime #1365: Cannot build using Podman (docker alternative)

arduino-man, 2022-10-11:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Cannot build using Podman

### What should happen instead?

Infinitime firmware should be built

### Reproduction steps

Pull infinitime docker container, go into the Infinitime cloned repo, create a folder called "build" and then run: 

podman run --rm -it --user root -v $(pwd):/sources infinitime-build /opt/build.sh pinetime-app

### More details?

user@Users-MacBook-Pro InfiniTime % podman run --rm -it --user root -v $(pwd):/sources infinitime-build /opt/build.sh pinetime-app
+ set -e
+ export TOOLS_DIR=/opt
+ TOOLS_DIR=/opt
+ export SOURCES_DIR=/sources
+ SOURCES_DIR=/sources
+ export BUILD_DIR=/sources/build
+ BUILD_DIR=/sources/build
+ export OUTPUT_DIR=/sources/build/output
+ OUTPUT_DIR=/sources/build/output
+ export BUILD_TYPE=Release
+ BUILD_TYPE=Release
+ export GCC_ARM_VER=gcc-arm-none-eabi-9-2020-q2-update
+ GCC_ARM_VER=gcc-arm-none-eabi-9-2020-q2-update
+ export NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
+ NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
++ uname -m
+ MACHINE=aarch64
+ [[ aarch64 == \a\r\m\6\4 ]]
+ [[ false == \f\a\l\s\e ]]
+ main pinetime-app
+ local target=pinetime-app
+ mkdir -p /opt
+ [[ ! -d /opt/gcc-arm-none-eabi-9-2020-q2-update ]]
+ [[ ! -d /opt/nRF5_SDK_15.3.0_59ac345 ]]
+ [[ ! -d /opt/mcuboot ]]
+ mkdir -p /sources/build
+ CmakeGenerate
+ cd /sources/build
+ cmake -G 'Unix Makefiles' -DCMAKE_BUILD_TYPE=Release -DUSE_OPENOCD=1 -DARM_NONE_EABI_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update -DNRF5_SDK_PATH=/opt/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 /sources
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PROJECT_GIT_COMMIT_HASH_SUCCESS? 0

BUILD CONFIGURATION
-------------------
    * Mode : Release
    * Version : 1.10.0
    * Toolchain : /opt/gcc-arm-none-eabi-9-2020-q2-update
    * GitRef(S) : 6dd67eb5
    * NRF52 SDK : /opt/nRF5_SDK_15.3.0_59ac345
    * Target device : PINETIME
    * Debug pins : Disabled
    * Build DFU (using adafruit-nrfutil) : Enabled
    * Build resources : Disabled
-- Using LV_FONT_CONV-NOTFOUND to generate font files
You have called ADD_LIBRARY for library infinitime_fonts without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error at src/CMakeLists.txt:878 (add_library):
  Cannot find source file:

    libs/lvgl/lvgl.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at src/CMakeLists.txt:890 (add_library):
  Cannot find source file:

    libs/QCBOR/src/ieee754.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at src/CMakeLists.txt:909 (add_library):
  Cannot find source file:

    libs/littlefs/lfs_util.h

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error: Cannot determine link language for target "lvgl".
CMake Error: CMake can not determine linker language for target: lvgl
CMake Error: Cannot determine link language for target "littlefs".
CMake Error: CMake can not determine linker language for target: littlefs
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    USE_OPENOCD

-- Build files have been written to: /sources/build
+ cmake -L -N .
-- Cache values
CMAKE_BUILD_TYPE:STRING=Release
CMAKE_INSTALL_PREFIX:PATH=/usr/local
LV_FONT_CONV:FILEPATH=LV_FONT_CONV-NOTFOUND
TARGET_DEVICE:STRING=PINETIME
+ CmakeBuild pinetime-app
+ local target=pinetime-app
+ [[ -n pinetime-app ]]
+ target='--target pinetime-app'
++ nproc
+ cmake --build /sources/build --config Release --target pinetime-app -- -j1
Scanning dependencies of target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  0%] Generating jetbrains_mono_extrabold_compressed.c
  File "/sources/src/displayapp/fonts/generate.py", line 21
    def gen_lvconv_line(lv_font_conv: str, dest: str, size: int, bpp: int, sources: typing.List[Source], compress:bool=False):
                                    ^
SyntaxError: invalid syntax
src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/build.make:60: recipe for target 'src/displayapp/fonts/jetbrains_mono_extrabold_compressed.c' failed
make[3]: *** [src/displayapp/fonts/jetbrains_mono_extrabold_compressed.c] Error 1
CMakeFiles/Makefile2:798: recipe for target 'src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/all' failed
make[2]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/all] Error 2
CMakeFiles/Makefile2:362: recipe for target 'src/CMakeFiles/pinetime-app.dir/rule' failed
make[1]: *** [src/CMakeFiles/pinetime-app.dir/rule] Error 2
Makefile:196: recipe for target 'pinetime-app' failed
make: *** [pinetime-app] Error 2
+ return 1
+ BUILD_RESULT=1
+ '[' '' '!=' true -a 1 == 0 ']'

### Version

Latest (cloned from https://github.com/InfiniTimeOrg/InfiniTime)

FintasticMan, 2022-10-12:
It seems like you didn't clone the submodules along with the InfiniTime repository. Could you try it again after running `git submodule update --init --recursive` in the InfiniTime repo?

arduino-man, 2022-10-12:
Thank you.

I ran: `git submodule update --init --recursive`

And my output is now the following:

```
user@Users-MacBook-Pro InfiniTime % podman run --rm -it --user root -v $(pwd):/sources infinitime-build /opt/build.sh pinetime-app
+ set -e
+ export TOOLS_DIR=/opt
+ TOOLS_DIR=/opt
+ export SOURCES_DIR=/sources
+ SOURCES_DIR=/sources
+ export BUILD_DIR=/sources/build
+ BUILD_DIR=/sources/build
+ export OUTPUT_DIR=/sources/build/output
+ OUTPUT_DIR=/sources/build/output
+ export BUILD_TYPE=Release
+ BUILD_TYPE=Release
+ export GCC_ARM_VER=gcc-arm-none-eabi-9-2020-q2-update
+ GCC_ARM_VER=gcc-arm-none-eabi-9-2020-q2-update
+ export NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
+ NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
++ uname -m
+ MACHINE=aarch64
+ [[ aarch64 == \a\r\m\6\4 ]]
+ [[ false == \f\a\l\s\e ]]
+ main pinetime-app
+ local target=pinetime-app
+ mkdir -p /opt
+ [[ ! -d /opt/gcc-arm-none-eabi-9-2020-q2-update ]]
+ [[ ! -d /opt/nRF5_SDK_15.3.0_59ac345 ]]
+ [[ ! -d /opt/mcuboot ]]
+ mkdir -p /sources/build
+ CmakeGenerate
+ cd /sources/build
+ cmake -G 'Unix Makefiles' -DCMAKE_BUILD_TYPE=Release -DUSE_OPENOCD=1 -DARM_NONE_EABI_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-9-2020-q2-update -DNRF5_SDK_PATH=/opt/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 /sources
PROJECT_GIT_COMMIT_HASH_SUCCESS? 0

BUILD CONFIGURATION
-------------------
    * Mode : Release
    * Version : 1.10.0
    * Toolchain : /opt/gcc-arm-none-eabi-9-2020-q2-update
    * GitRef(S) : 6dd67eb5
    * NRF52 SDK : /opt/nRF5_SDK_15.3.0_59ac345
    * Target device : PINETIME
    * Debug pins : Disabled
    * Build DFU (using adafruit-nrfutil) : Enabled
    * Build resources : Disabled
-- Using LV_FONT_CONV-NOTFOUND to generate font files
You have called ADD_LIBRARY for library infinitime_fonts without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
-- Generating done
-- Build files have been written to: /sources/build
+ cmake -L -N .
-- Cache values
CMAKE_BUILD_TYPE:STRING=Release
CMAKE_INSTALL_PREFIX:PATH=/usr/local
LV_FONT_CONV:FILEPATH=LV_FONT_CONV-NOTFOUND
TARGET_DEVICE:STRING=PINETIME
+ CmakeBuild pinetime-app
+ local target=pinetime-app
+ [[ -n pinetime-app ]]
+ target='--target pinetime-app'
++ nproc
+ cmake --build /sources/build --config Release --target pinetime-app -- -j1
make: Warning: File 'Makefile' has modification time 37503 s in the future
make[1]: Warning: File 'CMakeFiles/Makefile2' has modification time 37506 s in the future
make[2]: Warning: File 'CMakeFiles/Makefile2' has modification time 37506 s in the future
make[3]: Warning: File 'src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/progress.make' has modification time 37506 s in the future
make[3]: warning:  Clock skew detected.  Your build may be incomplete.
make[3]: Warning: File 'src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/progress.make' has modification time 37506 s in the future
[  1%] Generating jetbrains_mono_extrabold_compressed.c
  File "/sources/src/displayapp/fonts/generate.py", line 21
    def gen_lvconv_line(lv_font_conv: str, dest: str, size: int, bpp: int, sources: typing.List[Source], compress:bool=False):
                                    ^
SyntaxError: invalid syntax
src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/build.make:60: recipe for target 'src/displayapp/fonts/jetbrains_mono_extrabold_compressed.c' failed
make[3]: *** [src/displayapp/fonts/jetbrains_mono_extrabold_compressed.c] Error 1
CMakeFiles/Makefile2:798: recipe for target 'src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/all' failed
make[2]: *** [src/displayapp/fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/all] Error 2
CMakeFiles/Makefile2:362: recipe for target 'src/CMakeFiles/pinetime-app.dir/rule' failed
make[1]: *** [src/CMakeFiles/pinetime-app.dir/rule] Error 2
Makefile:196: recipe for target 'pinetime-app' failed
make: *** [pinetime-app] Error 2
+ return 1
+ BUILD_RESULT=1
+ '[' '' '!=' true -a 1 == 0 ']'
user@Users-MacBook-Pro InfiniTime % 
```

minacode, 2022-10-12:
That syntax error looks like Python 2 or maybe an older Python 3 without type hints.

arduino-man, 2022-10-12:
What can I do to fix? Is this a problem with my Mac defaulting to Python2 or with the Python version that comes with the container?

arduino-man, 2022-10-12:
Maybe it is an issue with my Mac being ARM but since the container *is* running, I think that is not the issue. I am too new to containers to diagnose the problem on my own unfortunately.

minacode, 2022-10-13:
Ok, so I tried this for a short time because you made me curious. But you have to give me more detailed reproduction steps because I am not using podman.

Or you do what [the introduction docs](https://docs.podman.io/en/latest/Introduction.html) say and run
```
podman run -it image_name
```
where `image_name` is the name of your image. You can get this with
```
podman images
```
if I get that right.
The above command should open a shell inside the container and there you can run
`python3` or `python3 -V` to run a python shell or get the version.
Lets see what we can find there.

arduino-man, 2022-10-13:
I think that the container is preconfigured to run some script at boot time because I pulled a brand new image, ran it and I get the following output: 

```
user@Users-MacBook-Pro ~ % podman run -it docker.io/infinitime/infinitime-build
Trying to pull docker.io/infinitime/infinitime-build:latest...
Getting image source signatures
Copying blob sha256:85f0d0fcd8735c107fb21d1730def84d34c3adc8a80e46c723750e42afec33ea
Copying blob sha256:00f50047d6061c27e70588a5aab89adada756e87d782a6c6bd08b4139eb8ea10
Copying blob sha256:f8e5d28e03f15b2f90d91cc983e2592420b6bbacd79b41337dab14a134ef4835
Copying blob sha256:0c336a534c78e8c221b4defeedb602355ea622f3c9c2117361a91f65e87784ef
Copying blob sha256:f7be0bf44b2d016b1f7a7bdbee0457715977815ed8a6378b0eacab7ffd33fec6
Copying blob sha256:766e439435a2ee696d84ce0e60dc74e72459229ea7fa272ca465af1260bdcff4
Copying blob sha256:9dff6b19716209b370caf6d152478bc44e0926ba0c87cf8923f8960d5fc29c06
Copying blob sha256:9621cf001bfec951388969faacdd8271c216bda0bc77283121200b30f10745b1
Copying blob sha256:50558273dbeb42edaae3f68f9c9932aceeac400f55a927d095a97da0b4732f1f
Copying blob sha256:75c7561f9a59469850e368b7a718b7397ea0a7774093f98b50fc7261fb437dcd
Copying blob sha256:39057f8ce7460c757a395c686ec829471e38817d9cf2edc3d76249b11f6f846a
Copying blob sha256:aab311d2eb210a77763e64baec34e3e91f7040cb42013c349085aed177349c52
Copying blob sha256:4da5a3715b954e59d0dda9212f529e50d957c2a7ade9c56dc2858b6abad38947
Copying config sha256:01accbcd89858afa0f8899a591858749701effa1cc086e122a11056d6636fa7e
Writing manifest to image destination
Storing signatures
+ set -e
+ export TOOLS_DIR=/opt
+ TOOLS_DIR=/opt
+ export SOURCES_DIR=/sources
+ SOURCES_DIR=/sources
+ export BUILD_DIR=/sources/build
+ BUILD_DIR=/sources/build
+ export OUTPUT_DIR=/sources/build/output
+ OUTPUT_DIR=/sources/build/output
+ export NPM_DIR=/sources/build/npm
+ NPM_DIR=/sources/build/npm
+ export npm_config_cache=/sources/build/npm
+ npm_config_cache=/sources/build/npm
+ export BUILD_TYPE=Release
+ BUILD_TYPE=Release
+ export GCC_ARM_VER=10.3-2021.10
+ GCC_ARM_VER=10.3-2021.10
+ export NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
+ NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
++ uname -m
+ MACHINE=aarch64
+ [[ aarch64 == \a\r\m\6\4 ]]
+ export GCC_ARM_PATH=gcc-arm-none-eabi-10.3-2021.10
+ GCC_ARM_PATH=gcc-arm-none-eabi-10.3-2021.10
+ [[ false == \f\a\l\s\e ]]
+ main
+ local target=
+ mkdir -p /opt
+ [[ ! -d /opt/gcc-arm-none-eabi-10.3-2021.10 ]]
+ [[ ! -d /opt/nRF5_SDK_15.3.0_59ac345 ]]
+ [[ ! -d /opt/mcuboot ]]
+ mkdir -p /sources/build
+ CmakeGenerate
+ cmake -G 'Unix Makefiles' -S /sources -B /sources/build -DCMAKE_BUILD_TYPE=Release -DARM_NONE_EABI_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-10.3-2021.10 -DNRF5_SDK_PATH=/opt/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 -DBUILD_RESOURCES=1
CMake Error: The source directory "/sources" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
+ CmakeBuild
+ local target=
+ [[ -n '' ]]
++ nproc
+ cmake --build /sources/build --config Release -- -j1
Error: could not load cache
+ return 1
+ BUILD_RESULT=1
+ '[' '' '!=' true -a 1 == 0 ']'
user@Users-MacBook-Pro ~ % 

```

FintasticMan, 2022-10-13:
Well, I've just installed podman and tried it out on my Linux machine, and it works fine using the command `podman run --rm -it -v "${PWD}:/sources" --user "$(id -u):$(id -g)" docker.io/infinitime/infinitime-build /opt/build.sh pinetime-app`. Based some quick searches, that should also work on macOS. It would be a good idea to remove the `build` directory in the InfiniTime repo, so you start with a clean slate.

arduino-man, 2022-10-13:
> Well, I've just installed podman and tried it out on my Linux machine, and it works fine using the command `podman run --rm -it -v "${PWD}:/sources" --user "$(id -u):$(id -g)" [...]

Thanks for going as far as trying it on your own computer. If it's not too much to ask can you give me the exact steps that you followed? In my case, I am simply cloning the repo, going into the repo directory and creating a new directory called build, and finally I run the command that you just shared to no avail.

FintasticMan, 2022-10-13:
All I did was clone the repo, clone the submodules with `git submodule update --init --recursive`, then run the podman command from the repo directory.

arduino-man, 2022-10-13:
Did you have to create a "build" directory inside of the InfiniTime directory?

This is the output that I got when cloning the repo, cloning the submodules and then running the Podman command.

```
user@Users-MacBook-Pro containers % git clone https://github.com/InfiniTimeOrg/InfiniTime.git                                                                           
Cloning into 'InfiniTime'...
remote: Enumerating objects: 17653, done.
remote: Counting objects: 100% (343/343), done.
remote: Compressing objects: 100% (202/202), done.
remote: Total 17653 (delta 226), reused 217 (delta 141), pack-reused 17310
Receiving objects: 100% (17653/17653), 45.32 MiB | 10.81 MiB/s, done.
Resolving deltas: 100% (12243/12243), done.
user@Users-MacBook-Pro containers % cd InfiniTime 
user@Users-MacBook-Pro InfiniTime % 
user@Users-MacBook-Pro InfiniTime % git submodule update --init --recursive
Submodule 'src/libs/QCBOR' (https://github.com/laurencelundblade/QCBOR.git) registered for path 'src/libs/QCBOR'
Submodule 'src/libs/date' (https://github.com/HowardHinnant/date.git) registered for path 'src/libs/date'
Submodule 'src/libs/littlefs' (https://github.com/littlefs-project/littlefs.git) registered for path 'src/libs/littlefs'
Submodule 'src/libs/lvgl' (https://github.com/joaquimorg/lvgl.git) registered for path 'src/libs/lvgl'
Cloning into '/Users/user/Documents/development/containers/InfiniTime/src/libs/QCBOR'...
Cloning into '/Users/user/Documents/development/containers/InfiniTime/src/libs/date'...
Cloning into '/Users/user/Documents/development/containers/InfiniTime/src/libs/littlefs'...
Cloning into '/Users/user/Documents/development/containers/InfiniTime/src/libs/lvgl'...
Submodule path 'src/libs/QCBOR': checked out '9e2f70804393823cc6d16f9f1035ef7223faca04'
Submodule path 'src/libs/date': checked out '9502bc27a3b32fa5759ea82a76c22551620ebcd2'
Submodule path 'src/libs/littlefs': checked out 'ead50807f1ca3fdf2da00b77a0ce02651ded2d13'
Submodule path 'src/libs/lvgl': checked out '23430cf20e32294549fff9b2879a9466dacc19bb'
user@Users-MacBook-Pro InfiniTime % 
user@Users-MacBook-Pro InfiniTime % 
user@Users-MacBook-Pro InfiniTime % podman run --rm -it -v "${PWD}:/sources" --user "$(id -u):$(id -g)" docker.io/infinitime/infinitime-build /opt/build.sh pinetime-app
+ set -e
+ export TOOLS_DIR=/opt
+ TOOLS_DIR=/opt
+ export SOURCES_DIR=/sources
+ SOURCES_DIR=/sources
+ export BUILD_DIR=/sources/build
+ BUILD_DIR=/sources/build
+ export OUTPUT_DIR=/sources/build/output
+ OUTPUT_DIR=/sources/build/output
+ export NPM_DIR=/sources/build/npm
+ NPM_DIR=/sources/build/npm
+ export npm_config_cache=/sources/build/npm
+ npm_config_cache=/sources/build/npm
+ export BUILD_TYPE=Release
+ BUILD_TYPE=Release
+ export GCC_ARM_VER=10.3-2021.10
+ GCC_ARM_VER=10.3-2021.10
+ export NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
+ NRF_SDK_VER=nRF5_SDK_15.3.0_59ac345
++ uname -m
+ MACHINE=aarch64
+ [[ aarch64 == \a\r\m\6\4 ]]
+ export GCC_ARM_PATH=gcc-arm-none-eabi-10.3-2021.10
+ GCC_ARM_PATH=gcc-arm-none-eabi-10.3-2021.10
+ [[ false == \f\a\l\s\e ]]
+ main pinetime-app
+ local target=pinetime-app
+ mkdir -p /opt
+ [[ ! -d /opt/gcc-arm-none-eabi-10.3-2021.10 ]]
+ [[ ! -d /opt/nRF5_SDK_15.3.0_59ac345 ]]
+ [[ ! -d /opt/mcuboot ]]
+ mkdir -p /sources/build
mkdir: cannot create directory ‘/sources/build’: Permission denied
+ CmakeGenerate
+ cmake -G 'Unix Makefiles' -S /sources -B /sources/build -DCMAKE_BUILD_TYPE=Release -DARM_NONE_EABI_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi-10.3-2021.10 -DNRF5_SDK_PATH=/opt/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1 -DBUILD_RESOURCES=1
CMake Error at /usr/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:181 (file):
  file failed to open for writing (No such file or directory):

    /sources/build/CMakeFiles/CMakeOutput.log
[... 495 lines cut ...]
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:5 (project)

-- Configuring incomplete, errors occurred!
CMake Error: Cannot open file for write: /sources/build/CMakeCache.txt.tmp8200a
CMake Error: : System Error: No such file or directory
CMake Error: Unable to open cache file for save. /sources/build/CMakeCache.txt
CMake Error: : System Error: No such file or directory
+ CmakeBuild pinetime-app
+ local target=pinetime-app
+ [[ -n pinetime-app ]]
+ target='--target pinetime-app'
++ nproc
+ cmake --build /sources/build --config Release --target pinetime-app -- -j1
Error: /sources/build is not a directory
+ return 1
+ BUILD_RESULT=1
+ '[' '' '!=' true -a 1 == 0 ']'
user@Users-MacBook-Pro InfiniTime % 
```

arduino-man, 2022-10-13:
I changed `--user "$(id -u):$(id -g)"` to `--user root` and now things seem to be working! Currently compiling (10% progress)

arduino-man, 2022-10-13:
Marking as resolved as I am now able to compile the project.

closed 2022-10-13 by arduino-man