InfiniTimeOrg/InfiniTime #1371: Sometimes I can't sleep my watch

wmertens, 2022-10-16:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I click the button to turn off the screen but it immediately wakes up again

### What should happen instead?

It should go to sleep

### Reproduction steps

It's not always happening, but I only have shake-wake enabled.

Sometimes when I'm at the clock and I click the button, the watch goes to sleep and immediately wakes up again. The workaround is to wait for the timeout so that it goes to sleep by itself.

### Version

0.10.0

### Companion app

GadgetBridge

minacode, 2022-10-16:
Could it be that you shake the watch while pressing the button?
Please try again with disabled shake wake. 

Also I guess you mean version 1.10?

Riksu9000, 2022-10-16:
Duplicate of #927

closed 2022-10-16 by Riksu9000