InfiniTimeOrg/InfiniTime #1372: Can't install resources

Vistaus, 2022-10-16:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I tried to install the resources with itd, but there's no way to do it.

### Reproduction steps

I tried to install the resources with itd as the documentation suggested, but there's no way to do it.

```
itctl res load infinitime-resources-1.11.0.zip
No help topic for 'res'
```

is the output I get. I wasn't sure if I should report this to itd, since it's your documentation that shows this command as working.

### Version

1.11.0

### Companion app

itd git master

FintasticMan, 2022-10-16:
You need to use the `resource-loading` branch of itd to do that.

Avamander, 2022-10-16:
Also not an InfiniTime issue.

closed 2022-10-16 by Avamander

Vistaus, 2022-10-16:
> You need to use the `resource-loading` branch of itd to do that.

I'm using that branch. 

> Also not an InfiniTime issue.

Not entirely, but you do provide installation instructions. If you provide instructions, then they should be working. If not, then they should be removed from the documentation.

Vistaus, 2022-10-16:
Alright, I recompiled itd a few times from that branch and now I got it mostly working. However, it still doesn't install the resources as it doesn't get past

```
./itctl res load ~/Downloads/infinitime-resources-1.11.0.zip
Removing example-of-obsolete-file.bin
```

Vistaus, 2022-10-17:
Could you, as the devs, at the very least re-test flashing the resources ZIP file from the Releases page to confirm whether something's wrong with the ZIP or an issue on my end?

Itai-Nelken, 2022-10-17:
ITD had a bug with the 1.11 resources zip.
V0.0.8 fixed it.
So install [itd 0.0.8](https://gitea.arsenm.dev/Arsen6331/itd/releases/tag/v0.0.8) and try again.

Vistaus, 2022-10-17:
> ITD had a bug with the 1.11 resources zip. V0.0.8 fixed it. So install [itd 0.0.8](https://gitea.arsenm.dev/Arsen6331/itd/releases/tag/v0.0.8) and try again.

That seemed to have worked. Thank you for the fix! :-)

-----------------

Still odd that it didn't work with Amazfish either, though. I still wonder if it has something to do with the resources file, or does Amazfish have the same "bug" as itd?

gelav, 2022-10-25:
https://github.com/InfiniTimeOrg/InfiniTime/issues/1378#issuecomment-1289887885