InfiniTimeOrg/InfiniTime #1374: Have timer go negative after it hits 0:00

calhix, 2022-10-16:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I'd like the timer to go negative so I know how long it has been since it hit 0:00

### Description

It would be nice that after the timer hits 0:00, a separate screen shows up with the timer counting negative to let you know how long it has been since the timer hit 0:00.

FintasticMan, 2022-10-16:
There is a PR that does that: #557, but that is severely outdated.

Riksu9000, 2022-12-29:
@calhix @emosenkis Would this still be useful if the timer continued vibrating, because then you probably wouldn't miss it anyway and I suspect it'd be annoying enough that it would be stopped immediately?

emosenkis, 2022-12-29:
I imagine that it's far from ideal for this use case and I'd be happy to do a follow-up that allows the user to silence the vibrations or just turn them off after 10 seconds but I'd prefer to keep the PR, which I think is strictly an improvement, focused in the hopes of merging it quickly.

Riksu9000, 2022-12-29:
I like the idea of having the ringing stop after ~10 seconds and keeping the overtime. I'll give the PR a review soon.