InfiniTimeOrg/InfiniTime #1376: 1.11 Watch face data corruption

danielrparks, 2022-10-17:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

After playing with different watch faces, the Casio 7-segment font was corrupted

### What should happen instead?

The font should continue to work no matter how I interact with the watch

### Reproduction steps

Not sure if it will be possible to reproduce this one.

I:

- Installed the resource pack using itd
- Tried all the different watch faces. They all worked.
- Went back to PineTimeStyle and played with the colors for a while.
- Waited several hours
- Played with the watch faces again. The Casio font was corrupted.

### More details?

This is a sealed watch, unfortunately, so I don't have swd access.

### Version

v1.11.0

### Companion app

Gadgetbridge 0.70.0, itd r199.1e072a3

danielrparks, 2022-10-17:
[image]

danielrparks, 2022-10-17:
I reset the watch and the face loaded directly, so the corruption was in-memory and not in the files themselves.

OskarSchamardin, 2022-10-21:
In my case all the custom fonts failed to load. Same fix, just swiping away and back fixes it. 

Some debugging ideas from a layman's perspective (not familiar with this codebase):

- Maybe force the watch to only show the screen after loading the fonts. 
- Compare the sequence of events what occur when waking the watch with a chime vs pressing the button (i.e what functions get called with button press and if the same ones get called with chime). I see no observable time difference when waking with button or chime.

(I tried to add a picture, but github decided to not work, will try again in an edit)

coledeck, 2022-10-22:
As mentioned in another comment of #1377 , I also see nothing in terms of visible assets (like the photo in https://github.com/InfiniTimeOrg/InfiniTime/issues/1377#issuecomment-1287387076) on the casio watchface specifically when it wakes for chime every hour. That migt be a separate issue because it seems to only happen durnign chime wakeup. Going to another app and back fixes it.

mark9064, 2025-06-28:
I believe this has been resolved by heap unification and chime fixes. Feel free to re-open / open a new ticket if this still happens

closed 2025-06-28 by mark9064