InfiniTimeOrg/InfiniTime #1377: Infineat watchface pinelogo dissapering and date cut off

ThatGuyInfinity, 2022-10-18:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I have set up chimes to wake my pinetime every hour. When that happens and the screen wakes up the pine logo (charging status) dissapears and date gets partialy cut off (it also changes font - it gets much thicker). When I swipe from the watchface for eg. from right to left and then swipe back, the watchface is back to normal.

### Reproduction steps

Set the Infinineat watchface and then chimes, when the chimes automatically wake up the screen you´ll see that the watchface looks as described above.

### More details?

[image]

### Version

v1.11.0

### Companion app

itd

JF002, 2022-10-19:
This is probably linked to https://github.com/InfiniTimeOrg/InfiniTime/issues/1376 : for whatever reason, LVGL couldn't allocate memory (RAM) to load the font. In this case, Infineat switches to a fallback mode that use the system font. The layout still need a bit of improvement in that case. 

As I said in #1376 : reproducing and diagnosing this issue. I hope that optimizing memory usage will help in that case.

OskarSchamardin, 2022-10-21:
[image]
In my case all the custom fonts failed to load with casio style layout. This happened when a chime (30min) wakes up the watch. Fixed by just swiping away and back.

Some debugging ideas from a layman's perspective (not familiar with this codebase):

- Maybe force the watch to only show the screen after loading the fonts. 
- Compare the sequence of events what occur when waking the watch with a chime vs pressing the button (i.e what functions get called with button press and if the same ones get called with chime). I see no observable time difference when waking with button or chime.

Boteium, 2022-12-03:
This happens when I tried to load a 9kb image onto my watchface.
Should this be happening ?

[image]

And if I load anything larger than 20kb, the watch crashed within a minute.

JF002, 2022-12-07:
@Boteium I don't think it's directly related to this issue. Would you mind opening a new [Disussion](https://github.com/InfiniTimeOrg/InfiniTime/discussions/) for this question?

nickalcock, 2023-07-27:
I saw this in 1.11: in 1.13, the Infineat side is fixed, but a related, worse bug has appeared: the same 'missing fonts on chime' that the Casio face has got has spread to Infineat as well. (Turning chimes off seems to fix it.)

I'd guess that waking the watch up via an alarm *and* rendering the face simultaneously hits problems, but it's not OOM: min free for me is 6312, which is fine.

minacode, 2023-08-02:
Have you tested the quick fix from #1411?

nickalcock, 2023-08-03:
Not yet: it'll be September before I'm near the necessary hardware to test it. I'll do so then.

mark9064, 2025-06-28:
This should be fixed due to chimes refactoring, feel free to re-open / open a new issue if not :)

closed 2025-06-28 by mark9064