InfiniTimeOrg/InfiniTime #1378: New watch faces not available after 1.11 DFU and resources load

mattklunder, 2022-10-18:
I applied the 1.11.0 DFU update and the resources update.  The output of the resource load is below:

itctl res load infinitime-resources-1.11.0.zip 
Uploading 7segments_115.bin 4928 / 4928 B |----------------------------| 100.00% 1m2s

After restarting the watch, the new watch faces are not available.  My concern is whether or not other updates may not be applied.  Is there anything else that needs to be done?

Enqueued, 2022-10-18:
I also just updated to firmware 1.11 via watchmate and the watch faces are also disabled for me.

InFerYes, 2022-10-18:
> 

Did you upload the resources via amazfish or tdi?

InFerYes, 2022-10-18:
> I also just updated to firmware 1.11 via watchmate and the watch faces are also disabled for me.

Did you upload the resources via amazfish or tdi?

mattklunder, 2022-10-18:
> GitHub](https://github.com/InfiniTimeOrg/InfiniTime/issues/1378#issuecomment-1283033802),
> auth/ALCW3G3WFKSI7ILTRAYQEILWD4KARANCNFSM6AAAAAARINETAM).
> thread.[image]Message

ITD

closed 2022-10-19 by Avamander

mattklunder, 2022-10-19:
i did install the resource file separately.  I installed the 1.11 DFU firmware update first without issues.  I then ran a second update separately, as noted in my original post, to install the resources file.  The release description you referenced didn't indicate I had to install anything else, separately or otherwise unless I wanted to do development.  Did I miss something?

Avamander, 2022-10-19:
Did you install the 1.11 update and forget to confirm the new version before resetting the watch?

InFerYes, 2022-10-19:
> i did install the resource file separately. I installed the 1.11 DFU firmware update first without issues. I then ran a second update separately, as noted in my original post, to install the [...]

I believe this ticket was closed prematurely as OP has a genuine issue, however others have not read the release notes properly and think they are experiencing a bug:

> To enable those new watch faces, you’ll need to install the new resource package. This package contains all the fonts and pictures needed by the watch faces and must be installed separately from the firmware. The installation procedure of the resources package is very similar to the update procedure of the firmware. **But, as this is a very new feature, not all companion apps support it.**
>
> - [ITD](https://gitea.arsenm.dev/Arsen6331/itd) integrated support for external resources in [version 0.0.8](https://gitea.arsenm.dev/Arsen6331/itd/releases/tag/v0.0.8).
>
> - It's also merged in [Amazfish](https://github.com/piggz/harbour-amazfish), and a new version with those changes should be published very soon.
>
> **Other companion apps will hopefully add support for this new feature soon !** Let's give the companion app developers time to integrate and release this new feature ;-)
>
> You’ll find more details about the installation procedure in [the documentation](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/gettingStarted/updating-software.md#updating-resources).

mattklunder, 2022-10-19:
> auth/ALCW3G2ADNEAHVZMXCMNTPLWD72BTANCNFSM6AAAAAARINETAM).

1.11 shows as validated on the watch.

mattklunder, 2022-10-20:
> What's the output of `itctl fs ls`?
> auth/ALCW3GZMVCTIPWSD7FQ554TWEGCNTANCNFSM6AAAAAARINETAM).

d 0.0 B .  

d 0.0 B ..  

d 0.0 B fonts  

\- 40 B settings.dat

mattklunder, 2022-10-20:
> Then I would say make sure you have the latest version of itd, and try
> running `itctl res load infinitime-resources-1.11.0.zip` again. There's no
> need to reboot after uploading the resources.

Thanks for the reply! I ran that command a couple of times. The first time,
the output was simply "Removing."  

  

The second time, the output looked like this:  

itctl res load infinitime-resources-1.11.0.zip  

Uploading 7segments_115.bin 4928 / 4928 B |----------------------------|
100.00% 1m2s  

  

I did upgrade itd to 0.0.8.  

  

After the resource load, the "itctl fs ls" command still showed this output:  

  

d 0.0 B .  

d 0.0 B ..  

d 0.0 B fonts  

\- 40 B settings.dat

mattklunder, 2022-10-20:
To confirm the version:  

  

itd --version  

0.0.8

prurigro, 2022-10-20:
Did you restart the itd service after updating?

mattklunder, 2022-10-20:
> auth/ALCW3G7WELZDQGEN4ZOXLX3WEGW3DANCNFSM6AAAAAARINETAM).

I don't recall when I restarted the service, but I have restarted the laptop
since, before I ran this last resource load.

prurigro, 2022-10-21:
Maybe try git master rather than 0.0.8 then? If it still doesn't work we'll at least know it's something other than itd

gelav, 2022-10-25:
The issue with Bluez, bt support in Linux. It should be version 5.65+ in order for ITD to flash resources properly. I found a solution in booting from LiveCD Ubuntu 22.04.1 and flashing from there.

Vistaus, 2022-10-25:
> The issue with Bluez, bt support in Linux. It should be version 5.65+ in order for ITD to flash resources properly. I found a solution in booting from LiveCD Ubuntu 22.04.1 and flashing from there.

I don't think that's the issue as my similar issue, #1372, was on openSUSE Tumbleweed with Bluez 5.65.

Avamander, 2022-10-25:
> I still think, based on my issue and this issue, that there's a bug in the resources file.

Very unlikely.

> I don't think Bluez is the culprit here as my similar issue, https://github.com/InfiniTimeOrg/InfiniTime/issues/1372, was on openSUSE Tumbleweed with Bluez 5.65.

Amazfish couldn't handle the MTU being reported incorrectly, there's a workaround for that now, but the underlying cause is not yet fully known. A bug in old bluez would be a reasonable explanation.

mattklunder, 2022-10-25:
> auth/ALCW3G2JUYPZ3G2QR2NK24TWE6RP3ANCNFSM6AAAAAARINETAM).

This issue has been fixed with an update to itd
(<https://gitea.arsenm.dev/Arsen6331/itd/issues/29#issuecomment-350>).

>> I still think, based on my issue and this issue, that there's a bug in the
resources file.  
>
>
> Very unlikely.  

>> I don't think Bluez is the culprit here as my similar issue,
[#1372](https://github.com/InfiniTimeOrg/InfiniTime/issues/1372), was on
openSUSE Tumbleweed with Bluez 5.65.

JF002, 2022-10-26:
> I still think, based on my issue and this issue, that there's a bug in the resources file.

I'm 100% positive that the resource file is fine. I've been able to install this file on  4 different PineTimes using 3 different computers, laptops and phones. And many other people also succeeded in installing the resource pack as well.

Please remember that the resource pack is a very new feature, for InfiniTime and also for companion apps! In fact, InfiniTime 1.11 was released even before the support for the feature was available in most of the companion apps. I thought I documented this in the release note and in the doc but maybe I was not clear enough about that. Sorry again for the confusion.

Since then ITD released 2 new versions ([0.0.8](https://gitea.arsenm.dev/Arsen6331/itd/releases/tag/v0.0.8) and [0.0.9](https://gitea.arsenm.dev/Arsen6331/itd/releases/tag/v0.0.9)) which improved the resource upload procedure. Amazfish has also release [version 2.1.0](https://github.com/piggz/harbour-amazfish/releases/tag/2.1.0) that supports the resource installation, and we have been busy [debugging issues related to this feature](https://github.com/piggz/harbour-amazfish/issues/246) as well.

In both apps, it was concluded that Bluez version 5.65 and up was needed for the feature to work correctly.

> but the devs don't really seem to care

We do care... but we also try to keep our backlog of issues clean and tidy this particular issue is very unlikely to be caused by a bug in InfiniTime . And I also spent most of my time debugging this kind of issue with other developers, which left not much time and energy to answer to many messages and comment on github...

Arsen6331, 2022-11-02:
> Amazfish couldn't handle the MTU being reported incorrectly, there's a workaround for that now, but the underlying cause is not yet fully known. A bug in old bluez would be a reasonable explanation.

@JF002 I actually do know the underlying cause. MTU was only added in 5.63, I believe (this actually caused issues for which I had to contribute a fix to the BlueZ library). Therefore, BlueZ versions before it returned zero, and since there's a filesystem request header, you had to subtract from that, which underflowed the unsigned integer.

JF002, 2022-11-05:
@Arsen6331 Thanks for the update about this issue! The issue and the solution were exactly the same for Amazfish running on SailfishOS (and other linux versions with older BlueZ) : the MTU property would return 0, which was interpreted as infinity. Amazfish would send a huge buffer to the watch, and InfiniTime ignored it as it was too big. Upgrading to Bluez 5.65 fixed the issue on SailfishOS.