InfiniTimeOrg/InfiniTime #1382: Should I submit PR's for apps that are a little niche?

Derek52, 2022-10-20:
Hi, I am working on some apps for my Pinetime to help me when gaming. I have already built 2, a life point counter for Magic The Gathering, and a life point counter for Yu-Gi-Oh. Next, I'm thinking about making a dice roll simulator. 

Anyway, I like my life counter apps a lot, and want to know if I can get them merged into upstream. To get these apps working I also built a couple of Counter widgets. I took the Counter that is used in the timer app, and wrote a couple of new versions, one is a 2 digit, one is a 4 digit. They may have uses for someone down the line. I'm not sure.

I think my life point counter apps have use for people. But, I don't think they are going to be used enough to make them apps enabled by default. But, there may be users who want to go through the code and enable them for their own watches. Also, if you merge them into upstream, I'll have a much easier time updating my own personal watch's build.

Basically, I want to know if there is interest in me submitting a couple of PR's that add apps to the watch, without automatically enabling them.

FintasticMan, 2022-10-20:
At the moment, that kind of app will not be merged because of space constraints, but I think it is a good idea to make pull requests anyway. This way you can get feedback on your app and its code, and when we have a way to load apps from the external flash, it might be merged sooner.

yusufmte, 2022-10-21:
As an aside, dice rolling app PR here https://github.com/InfiniTimeOrg/InfiniTime/pull/1326 in case you are interested :) Basic functionality in it is complete from my end and I've been using it on my pinetime. I'd be curious if it suits your needs / if it's similar to what you were thinking of.

minacode, 2022-10-21:
Well.. #1296.

Derek52, 2022-10-22:
@yusufmte I like your dice rolling app a lot. I like that it appears to change the font color when rolling to help you visualize that you have gotten a new result. I was trying to come up with a simple way to show the result had changed, when say, you get a 3 two times in a row. I guess I don't have to make a dice rolling app after all.

yusufmte, 2022-10-22:
@Derek52 Thanks -- glad you liked it!

By the way, IMO I agree with @FintasticMan about submitting the PR. Who knows, there may be more people interested in those specific uses than anticipated, and it wouldn't really affect the user experience for people who don't use it. Besides, the improvements to the Counter class could be generally useful. And the MTG counter would just be two 2-digit counters with a + and -, right? So it could perhaps be used as a general scorekeeping app for MTG but also for anything like ping pong, rock paper scissors, etc.?

RageGamerBoi, 2022-10-24:
People have been suggesting a score keeping app for ages, and it does feel like we need a basic one preinstalled…
It could literally just be the timer app with the secondary reset function made as the main button, and the timer bits ripped out.
Same goes for the dice app (as implemented in #1326 ).
You shouldn’t have to install basic apps on to the watch (ie. you should be able to use the watch almost entirely without Bluetooth), but I can see one specifically for a certain card game needing to be installed because it’s so niche.

JF002, 2022-10-26:
Feel free to open a PR with your app, of course! As you said, other developers and users might be interested with your work, you will probably receive feedback, ...

I can't promise anything about the merge of your changes in the main code base. As you can see, there are already many other PR waiting, and we unfortunately cannot integrate all of them at once.

Gernomaly, 2022-10-28:
Is there a good way to make Infinitime more modular so that it is easy to install an app you like and uninstall an app you dont. Maybe a bit like the ressources? (I have next to no clue how the pinetime works)

minacode, 2022-10-28:
There is an ok one if you are ok with compiling the firmware yourself and a harder one if not.

JF002, 2022-10-30:
> Is there a good way to make Infinitime more modular so that it is easy to install an app you like and uninstall an app you dont. Maybe a bit like the ressources? (I have next to no clue how the pinetime works)

There are currently 2 approaches : make InfiniTime more modular **at build time** by selecting which apps will be built in the firmware, and make InfiniTime more modular **at runtime** by allowing to install apps dynamically, using a system similar to the resources. Note that none of them has already been integrated (yet) in the project but a PR exists for the "build time modularity".

Supporting "dynamic apps" (apps that are not built into the firmware and that the user can install/remove at runtime) is definitely something we would like to add at some point, but we still have to figure out how to do this (implement an app loader, define a stable API between InfiniTime and the apps, find and allocate ram and flash memory to the apps,...).