InfiniTimeOrg/InfiniTime #1383: Music app UI update concept

Xeoy, 2022-10-20:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Add clock at top left and a time bar to the music app

### Description

Replace the current spinning disc in the music app and place a time bar with the current time and track length on the ends of the time bar. 

This could be used to seek to a track position by dragging the dot on the time bar. This same functionality can also be used to control volume of music. I'm not sure how of this can be implemented but removing the spinning disc and replacing it with a time bar might give a sleek look.

[image]

Also add clock to the music UI. It might be useful across all apps if it is possible to have a persistent clock on top left.

Avamander, 2022-10-20:
I tried a seek bar and it was way too finicky plus it isn't and can't be supported for most OSs and players.

The current small animation was added with the intention of being pushed cover art at some point in time.

Xeoy, 2022-10-21:
Thanks for the clarification. Would it be possible to have a clock at the top left on apps whenever it is possible? 

Album art would be dope! I'm not much of a coder but happy to help if there is a need for UI/icon designing!

RageGamerBoi, 2022-10-25:
The clock is (I believe) a widget that can be put on any screen. I’m not a developer so I don’t know exactly how, but it is a unified design that should be easy (enough) to implement.
Personally I think just the old music app with round buttons and a clock is good enough but this concept looks nice, maybe the slider could be used for volume?